carla-simulator/carla #4569: Problem with trying to run autopilot on OSM map

barasm-hita, 2021-08-23:
Hi, I'm issuing a problem that may be a bug or a mistake by me, so I'm not sure about but let me describe it to you.

**First of all these are details of my deployment of CARLA:**
CARLA 0.9.12 (downloaded ready built from releases) on Windows 10 21H1
NVIDIA RTX 2060, Intel Core i5 - 9400F, 32GB DDR4 2666 RAM and SSD Drive

**My Problem:** I started to follow instructions step by step from your official docs and I was trying to load OSM export into CARLA without converting it to OpenDrive format.  So I downloaded exactly the same .osm that has been used in docs from [Here](https://www.openstreetmap.org/export#map=15/41.5007/2.1228). Then, used method [C](https://carla.readthedocs.io/en/0.9.12/tuto_G_openstreetmap/#c-pass-osm-to-configpy) to pass the .osm file to config.py.
After all of these, I run manual_control.py from examples and by pressing button **P** on keyboard autopilot gets enabled. The problem comes here, that ego vehicle gets out of road (some times immediately).

**What I think:** I think that this problem is because of missing 3D objects and road signs (and lines) on OSM maps. My question is, how can I use autopilot without waypoints in OSM maps? Or, is it possible to add signs to roads on OSM maps?

I apologize for the lengthy explanation! 🙂
[image]

jackbart94, 2021-08-24:
The Traffic Manager now needs synchronous mode to work correctly. Please, add the ``--sync`` flag when executing ``manual_control.py`` if you intend to use it with the Traffic Manager.

barasm-hita, 2021-08-24:
> The Traffic Manager now needs synchronous mode to work correctly. Please, add the `--sync` flag when executing `manual_control.py` if you intend to use it with the Traffic Manager.

Thanks for your help. It solved my problem 🙏

closed 2021-08-24 by barasm-hita