carla-simulator/carla #4575: How can I add another camera (RGB and Depth) during the simulation?

rak7045, 2021-08-25:


rak7045, 2021-08-25:
I am trying to collect aerial images with bounding box information and class label with an RGB and depth camera. Now I want to add another camera to create a scene like drone is taking the pictures. For reference, I am attaching the picture of the view.  Now, I want to have another view on the side. How can I do that? Just add another camera or is there anyway to do that?
[image]

MattRowe18, 2021-11-16:
@rak7045 yes you can spawn multiple cameras to collect data on a scene, just repeat the process you use to spawn your first camera. Make sure you keep separate references in your script for each camera and ensure that you separate the output with different file paths. Refer to the scripts provided with CARLA to see how camera sensors are spawned.

closed 2021-11-16 by MattRowe18

husamhamu, 2022-02-05:
@rak7045 Hi, can you share your code to collect the data? thanks

husamhamu, 2022-03-04:
Thanks for your response @rak7045, I am actually interested in collecting a dataset like KITTI-dataset to train a model for monocular 3d object detection. There has been a work on gathering data like this in the following repo [https://github.com/Ozzyz/carla-data-export?utm_source=pocket_mylist](url) . But it does not work with the newer version of CARLA and I thought you are trying to do the same thing here with the new version of CARLA