carla-simulator/carla #4577: Uncatchable carla::client::TimeoutException on simulator crash

AreopagX, 2021-08-25:
CARLA version: 0.9.12
Platform/OS: Ubuntu 20.04
Problem you have experienced:
terminate called after throwing an instance of 'carla::client::TimeoutException'
  what():  time-out of 60000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
Aborted (core dumped)

What you expected to happen:
I expected a RuntimeException to be thrown and caught by a try-except clause covering the main function.

Steps to reproduce:

Run the code from the following file twice (it's a .txt file because GitHub won't allow uploading .py files).
[test.txt](https://github.com/carla-simulator/carla/files/7045635/test.txt)
- First run: without CARLA simulator running should give you the following output

> Caught exception!
Traceback (most recent call last):
  File "test.py", line 18, in main
    world = client.get_world()
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000

- Second run: start CARLA simulator before running the script and CTRL + C the simulator once it printed out "received image". This should yield the following output:
> received image
received image
received image
received image
received image
received image
received image
terminate called after throwing an instance of 'carla::client::TimeoutException'
  what():  time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
Aborted (core dumped)

I'd like to be able to restart the CARLA simulator when it crashes due to whatever reason by catching the timeout exception in order to run the simulation for some hours without the need for human supervision.

hh0rva1h, 2021-11-18:
Please label this as a bug / feature request. The culprit of this is traffic manager. As soon as traffic manager is initiated (which you do by setting the autopilot), there seems to be no way to terminate traffic manager in the API. This should be changed for exactly the reason mentioned in this report: Sometimes Carla crashes and one would like to catch that exception and handle a restart of Carla.

There are various requests spread across many different reports asking exactly for that:
- https://github.com/carla-simulator/carla/issues/3212#issuecomment-891480987

As a consequence people are forced to complicated multiprocessing workarounds, see for example:
- https://github.com/carla-simulator/carla/issues/2789#issuecomment-689619998

For me no workaround is possible, as running traffic manager in a separate process is not really possible in synchronous mode, since the client that ticks needs to hold traffic manager.

@bernatx @bernatx Please make this a high priority issue.

hh0rva1h, 2022-02-16:
@bernatx Would be amazing if you could give your opinion here.

MoritzNekolla, 2023-01-25:
> @bernatx Please provide some input here: would someone of the team be interested to improve this? If not then please directly close this, but it leaves a bad taste in user's mouths to let stale bot mass close issues without any dev input.

+1

OOOllie, 2024-06-21:
I am also having this issue, it is not possible for me to multiprocess it due to the nature of my application and communication to that process

M-Colley, 2026-10-26:
Proposed fix for the asynchronous local Traffic Manager worker termination: draft #10055, targeting `ue58-dev` and depending on #9999 for the thread-safe pending-exception channel.

The worker catches a timeout, stops, and forwards its message through Episode's existing pending-exception channel. Direct/synchronous calls still propagate exceptions normally; this does not raise a Python exception into a sleeping main thread. A replacement asynchronous worker resets its run flag.

The changed native source compiles and the component regressions pass 8/8 on Windows and Linux, including a real native RPC timeout caused by stopping a local protocol fixture. The faithful unprotected upstream worker entry aborts in the failure cases. Combined with the actual #9999 Episode RPC/streaming channel, 9/9 sanitizer tests pass. This is independently reviewed component validation; a complete local Traffic Manager running against Unreal and end-to-end simulator restart/recovery remain unrun.