carla-simulator/carla #4584: How to start server on colab?

schissmantics, 2021-08-28:
CARLA version: 0.9.12
Platform/OS: Ubuntu 18
Problem you have experienced: Unable to run server
What you expected to happen: 
Steps to reproduce:

According to package installation [docs][1], I install using the following:

    wget https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.12.tar.gz
    mkdir carla
    tar -zxvf CARLA_0.9.12.tar.gz -C carla
    cd carla
    ./ImportAssets.sh
    pip install pygame
    pip install PythonAPI/carla/dist/carla-0.9.12-cp37-cp37m-manylinux_2_27_x86_64.whl

Then I run using:

    sudo apt-get install libomp5 
    ./CarlaUE4.sh

I get an error:

    Refusing to run with the root privileges.

So, I create a new user

    adduser xyz
    sudo -u xyz ./CarlaUE4.sh

I get the following error:

    chmod: changing permissions of '/content/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping': Operation not permitted
    4.26.2-0+++UE4+Release-4.26 522 0
    Disabling core dumps.
    sh: 1: xdg-user-dir: not found
    error: XDG_RUNTIME_DIR not set in the environment.
    error: XDG_RUNTIME_DIR not set in the environment.
    error: XDG_RUNTIME_DIR not set in the environment.

Here's a [link][2] to the notebook

  [1]: https://carla.readthedocs.io/en/0.9.12/start_quickstart/#b-package-installation
  [2]: https://colab.research.google.com/drive/1fgDCgEhuuwv1HA4OYCWi9BgHWPuu8Riq?usp=sharing

qhaas, 2021-08-31:
> chmod: changing permissions of '/content/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping': Operation not permitted
4.26.2-0+++UE4+Release-4.26 522 0

Harmless warning, see Issue #3663
> sh: 1: xdg-user-dir: not found

Harmless warning, see Issue #3514

Looks like you are trying to run headless, xorg-less.  Try using vulkan with headless rendering mode enabled, per [Issue #4498](https://github.com/carla-simulator/carla/issues/4498#issuecomment-898644166):
```
unset SDL_VIDEODRIVER

./CarlaUE4.sh -vulkan -RenderOffscreen -nosound
```

schissmantics, 2021-08-31:
Thanks, I'll try again and let you know if the problem is gone.

schissmantics, 2021-08-31:
It's not working on colab, I get the same error. Please find the notebook [here](https://colab.research.google.com/drive/1bL4Dd6i7jsHTkrCtHWAPMBGszR8zC9iX?usp=sharing)

closed 2022-05-02 by stale[bot]

AOOOOOA, 2023-11-24:
> #3514

hi have u solved this?