carla-simulator/carla #4585: Modify default camera

zjh6252007, 2021-08-29:
Hello everyone
I'm now doing the multiply driving test. I found the fps are lower after the player joins the game. The more players join the low fps I get. I checked it in Unreal and I found it's because of the scene capture camera. I want to replace all of the scene capture cameras with the default UE4 camera. Is it possible for me to do that? What should I do if I can？

MattRowe18, 2021-11-18:
@Axel1092 could you answer this please

Axel1092, 2021-11-18:
Hi @zjh6252007, I am not sure what you mean with the multiple driving test, I guess you spawn a few vehicles with manual control maybe? In any case the scene capture camera cannot be replaced with the unreal camera. A scene capture camera is spawned when you create a camera sensor and renders the scene from that point of view and sends the image to the client.

closed 2022-04-30 by stale[bot]