carla-simulator/carla #4587: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found

ItsXianqi, 2021-08-30:
**CARLA version**: 0.9.12
**Platform/OS: Ubuntu**: 16.04
Hello dear developers, thanks for your work.
I'm trying to install package edition on Ubuntu 16.04, when i excuted
`./CarlaUE4.sh`
I met the error message
`/home/hexianqi/opt/CARLA_0.9.12/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: /lib/x86_64-linux-gnu/libm.so.6: version GLIBC_2.27 not found (required by /home/hexianqi/opt/CARLA_0.9.12/CarlaUE4/Binaries/Linux/../../Plugins/Carla/CarlaDependencies/lib/libChronoEngine.so)
/home/hexianqi/opt/CARLA_0.9.12/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: /lib/x86_64-linux-gnu/libm.so.6: version GLIBC_2.27 not found (required by /home/hexianqi/opt/CARLA_0.9.12/CarlaUE4/Binaries/Linux/../../Plugins/Carla/CarlaDependencies/lib/libChronoEngine_vehicle.so)`

It seems carla 0.9.12 requires GLIBC_2.27, and Ubuntu 16.04 is GLIBC_2.23. Because some other packages require Ubuntu 16.04,  I don't want to upgrade to Ubuntu 18.04. How can I solve the problem?

Thanks and best wishes.

qhaas, 2021-08-31:
Given the [era of 'free' Ubuntu 16.04 updates has ended](https://ubuntu.com/about/release-cycle), doubtful the upstream team is testing new releases against it.  You can likely work-around glibc issues using docker or singularity containers.

ItsXianqi, 2021-09-03:
> Given the [era of 'free' Ubuntu 16.04 updates has ended](https://ubuntu.com/about/release-cycle), doubtful the upstream team is testing new releases against it. You can likely work-around glibc issues using docker or singularity containers.

Got it, thanks

closed 2021-11-18 by MattRowe18

zhaoyongqi2022, 2022-07-14:
Hi, I encounter the same issue. Did you successfully solve this problem? Thank you in advance!

ItsXianqi, 2022-07-15:
> Hi, I encounter the same issue. Did you successfully solve this problem? Thank you in advance!

I changed to Ubuntu 20.04, sorry didn't fix it.

zhaoyongqi2022, 2022-07-19:
> > Hi, I encounter the same issue. Did you successfully solve this problem? Thank you in advance!
> 
> I changed to Ubuntu 20.04, sorry didn't fix it.

Hi, sorry to bother you again. Did you mean that you didn't solve the problem even after updating the system to 20.04?

ItsXianqi, 2022-07-19:
> > > Hi, I encounter the same issue. Did you successfully solve this problem? Thank you in advance!
> > 
> > 
> > I changed to Ubuntu 20.04, sorry didn't fix it.
> 
> Hi, sorry to bother you again. Did you mean that you didn't solve the problem even after updating the system to 20.04?

No, I change to 20.04 and it is fixed without extra modification

zhaoyongqi2022, 2022-07-19:
> No, I change to 20.04 and it is fixed without extra modification

Thanks for your reply!