carla-simulator/carla #4588: Server doesn't start or fails silently

schissmantics, 2021-08-30:
CARLA version: 0.9.12
Platform/OS: ubuntu 18
Problem you have experienced: server doesn't start
What you expected to happen: it should start
Steps to reproduce:

    >>> DISPLAY= carla/CarlaUE4.sh -opengl -carla-port=2000
    4.26.2-0+++UE4+Release-4.26 522 0
    Disabling core dumps.
    >>> lsof -i tcp:2000
    >>>

Nothing is displayed, which means server doesn't work.

qhaas, 2021-08-31:
The 'desktop' opengl renderer was removed in UE 4.26.x, see [Issue #4560](https://github.com/carla-simulator/carla/issues/4560#issuecomment-904943030)

qhaas, 2021-08-31:
Try the advice in [Issue #4584](https://github.com/carla-simulator/carla/issues/4584#issuecomment-909333606) for headless / xorg-less rendering, assuming the VM's GPU is working properly (e.g. `nvidia-smi -L` returns):
```
unset SDL_VIDEODRIVER
./CarlaUE4.sh -vulkan -RenderOffscreen -nosound
```

schissmantics, 2021-08-31:
I just tried your suggestion on gcp vm as well as on colab in [4584](https://github.com/carla-simulator/carla/issues/4584#issuecomment-909333606), and it doesn't work for either.

schissmantics, 2021-08-31:
I don't know, i've been trying to make it work on all kinds of possible devices, my mac included. I know it doesn't work on mac but i'm frustrated enough to try. In fact I've been trying to get any car simulator (carla, airsim, deepdrive, svl, ... you name it) to work anywhere, and all I've been getting is pain in my ass.

closed 2022-04-29 by stale[bot]

Djj646, 2022-08-15:
meeting the same issue, hope someone can land a hand

Ericwonne, 2023-02-24:
same issue, finished the download and extract, then failed the very first time, output:

```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Invalid MIT-MAGIC-COOKIE-1 keyInvalid MIT-MAGIC-COOKIE-1 keyInvalid MIT-MAGIC-COOKIE-1 key(base) 
```

Mariusmarten, 2024-08-26:
I face the same issue after updating my nvidia drivers: no additional output or logs are displayed.