carla-simulator/carla #4590: Can't run Carla due to missing GlobalShaderCache-PCD3D_SM5.bin and cooked/uncooked issue

Yebihashu, 2021-08-31:
CARLA version: 
Built version of 9.12
Building Carla was done exactly like instructed at : https://carla.readthedocs.io/en/0.9.12/build_windows with python 3.8.10

Platform/OS: 
Windows 10

Problem you have experienced: 
Running built CarlaUE4.exe shows the follow message and stops.
[image]

What you expected to happen:
According to https://carla.readthedocs.io/en/stable/running_simulator_standalone/ , running CarlaUE4.exe should launches the simulator window in full-screen, and you should be able now to drive around the city using the WASD keys, and Q for toggling reverse gear.

It was asked before in the forum but I didn't see a solution.
Also looking at https://forums.unrealengine.com/t/globalshadercache-pc3d_sm5-bin-is-missing/1784/2 there is advice to set build configuration as “DebugGame” , but the Carla instructions says that it should be "Development Editor"
[image]

Thanks

bernatx, 2021-09-07:
Did you get the assets? Maybe the Update.bat was failing at some point. I suggest to install the 7zip.org application to decompress big files, sometimes the Power shell is limited in terms of memory. The Update.bat script will use 7zip.exe if it is installed in its default path.

Yebihashu, 2021-09-09:
Thanks, 
Indeed, there was problem when calling Update.bat. So I did it manually to have all assets.

LinyeLi60, 2022-10-25:
Why closed!

myleskeller, 2023-02-08:
This may not be the solution to your particular issue, but mine turned out to be that I was launching the _wrong_ carlaUE4.exe.
I was launching the one located in:
`C:\path\to\carla\Unreal\CarlaUE4\Binaries\Win64`
but should have been launching the one in:
`C:\path\to\carla\Unreal\CarlaUE4\Saved\StagedBuilds\WindowsNoEditor`

ravivaghasiya1998, 2023-09-11:
Hello
i'm getting the same error when I try to run  CarlaUE4.exe from 'C:\path\to\carla\Unreal\CarlaUE4\Binaries\Win64'. I can not find CarlaUE4.exe in 'C:\path\to\carla\Unreal\CarlaUE4\Saved\StagedBuilds\WindowsNoEditor'. Inside the Saved folder there are only Autosaves,Collections,Config,Crashes and Logs folder. 

How can I get this file inside WindowsNoEditor directory ? or any solution to launch CarlaUE4.exe ?

alahi45, 2023-10-09:
> Hello i'm getting the same error when I try to run CarlaUE4.exe from 'C:\path\to\carla\Unreal\CarlaUE4\Binaries\Win64'. I can not find CarlaUE4.exe in [...]

I think you need to build the CarlaUE4.sln with Visual Studio file first for that. After that it'll appear in the ..\WindowsNoEditor .