carla-simulator/carla #4591: After compiling Carla server by "make launch" (windows) , the built UE4Editor.exe cannot work but only with path to CarlaUE4.uproject

Yebihashu, 2021-08-31:
CARLA version:
Built version of 9.12
Building Carla was done exactly like instructed at : https://carla.readthedocs.io/en/0.9.12/build_windows with python 3.8.10

Platform/OS:
Windows 10

Problem you have experienced:
Building UE4Editor.exe using VS2019 was ok. I could then run UE4Editor.exe from prompt window and the editor runs
[image]

But after compiling server using "make launch", I can't run UE4Editor.exe  anymore 
[image]

The only thing I can do is to run it with a path to CarlaUE4.uproject  
[image]

this will open the editor with Carla as shown
[image]

If I will compile UE4Editor.exe again in VS2019, then I will be able again,  to run UE4Editor.exe alone, 
but trying to run Carla with the editor failed

[image]

What you expected to have both possibilities:
To compile UE4Editor and Carla , and be able to:
(1) Use UE4Editor.exe  without Carla by running UE4Editor.exe  
(2) Use UE4Editor.exe  also with Carla by running UE4Editor.exe  [path to carla]/CarlaUE4.uproject  

Thanks

MattRowe18, 2021-11-18:
@bernat could you take a look at this please

closed 2022-04-29 by stale[bot]

hch-xmu, 2023-09-07:
Have you solve this,  I encounter this question aslo, Could you tell me how you do