carla-simulator/carla #4593: The running rate is abnormal after setting the fps

wjx-3510886, 2021-09-01:
CARLA version:0.9.11
Platform/OS:windows 10
Problem you have experienced:
After the FPS is set, the running rate of both the server and client is abnormal.(Pedestrians and vehicles are noticeably faster, and it takes less time for traffic lights to change)
What you expected to happen:
Run manual_control.py on the client at a fixed FPS (say 30) with valid synchronization mode to get the data
Steps to reproduce:1、Run the server     2、Run manual_control.py    3、spawn_npc.py --n 30   to add npcs     4、 config.py --fps 30 --delta-seconds 0.0333    5、spawn_npc.py --sync  
Other information (documentation you consulted, workarounds you tried):I want to manually control the vehicle and get some synchronous data (such as sensors and listening keyboard controls)， so I want to run manual_control.py in synchronized mode。So I reproduced the Steps to reproduce over and over again except for the first，Including the first step CarlaUE4.exe --benchmark --fps=30、CarlaUE4.exe --benchmark --fps=30 --delta-seconds=0.033,But it didn't do what I wanted, and instead it made the server and client run faster (obviously you can see pedestrians walking faster)Also, if I don't set sync mode, FPS, delta_seconds, and just add NPCS, running manual_control.py shows FPS is about 45 server and 60 client (up and down)
reference：
https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/

Thanks
-->

MattRowe18, 2021-11-18:
@jackbart94 could you look into this

jackbart94, 2021-11-18:
The --fps arg is definitely deprecated. Also, you need to run the ``sync`` script first, which will be the one doing the tick, and then all the remaining ones.

closed 2022-04-30 by stale[bot]