carla-simulator/carla #4595: Segmentation fault with generate_traffic.py

NiElAsS, 2021-09-01:
Hey, I encountered Segmentation fault when I tried to run the example "generate_traffic.py".

CARLA version: 0.9.12
Platform/OS: Ubuntu 20.04
Problem you have experienced: Got Segmentation fault (core dumped) with "python generate_traffic.py"
What you expected to happen: emm...generate traffic successfully?

Steps to reproduce:

make launch (worked fine)
click 'play'
python dynamic_weather.py (this worked fine)
ctrl+c (stop dynamic_weather.py)
python generate_traffic.py (here i got Segmentation fault (core dumped) )

jackbart94, 2021-09-02:
I've just tried it and everything worked fine for me. Could you run it again, maybe using ``gdb`` to identify what is causing the Segmentation Fault? I can guide you through it.

NiElAsS, 2021-09-02:
@jackbart94 Thanks for your reply!

I ran it again, and tried to find which part caused the problem. When I ran "world.apply_settings(settings)"(Line 187), my UE crushed.

I have to force quit my UE. And the UE terminal show 

```
[2021.09.02-14.59.55:405][637]LogTemp: Loaded OpenDrive file '/home/nielass/code/carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/Town10HD_Opt.xodr'
[2021.09.02-14.59.55:736][640]LogTemp: Loaded OpenDrive file '/home/nielass/code/carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/Town10HD_Opt.xodr'
/home/nielass/code/carla/Util/BuildTools/BuildCarlaUE4.sh: line 172: 29839 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 137

```

Could you help me with that?

jackbart94, 2021-09-02:
This doesn't say much. Please, do the following:
1. Before executing the ``generate_traffic.py``, in your terminal type ``gdb python``. This will launch gdb.
2. In the same terminal window, type ``run generate_traffic.py``. This will seg fault, at least in your machine.
3. Once seg faults, type ``bt``. This will print the backtrace of the error that caused the segmentation fault. If you can paste that backtrace here, it would help us quite a lot.

If anything is unclear, tell me!

NiElAsS, 2021-09-02:
@jackbart94 

Thanks for detailed instructions. 

'''
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
_PyObject_GC_Alloc (use_calloc=0, basicsize=88)
    at /home/builder/ktietz/cos6/ci_cos6/python-split_1622827272909/work/Modules/gcmodule.c:2226
2226	/home/builder/ktietz/cos6/ci_cos6/python-split_1622827272909/work/Modules/gcmodule.c: No such file or directory.
'''

jackbart94, 2021-09-02:
Looks like a dependency error. Make sure you've installed everything you need to run CARLA by following [this guide.](https://carla.readthedocs.io/en/0.9.12/build_linux/)

closed 2021-11-18 by MattRowe18

ZhouY-JustLL, 2022-07-22:
@jackbart94 

CARLA version: 0.9.13
Platform/OS: Ubuntu 18.04
python version: 3.9

I totally followed the building instruction in official carla documents, but met the same segmentation fault and same gdb information with generate_traffic.py. I tried some other scripts in PythonAPI/examples, and they all worked fine.

Then I find the segmentation fault was raise by `world.get_random_location_from_navigation()`. I hope this will give you more information about this problem.

If i couldn't solve the problem, is there any other way to randomly spawn a location for walkers?

JoseZF, 2022-08-09:
Check issue #5325. https://github.com/carla-simulator/carla/issues/5325#issue-1192688885