carla-simulator/carla #4605: Simulate Tesla's cameras

xia0long, 2021-09-03:
Hi there,

Maybe this problem has nothing to do with carla itself, What I want to ask is where I can find the calibration parameters of the 8 cameras in Tesla model 3? I want to simulate its camera in carla, and I think it would be OK if there were similar calibration parameters.

best regard

Domsall, 2021-09-07:
Tesla is always publishing data/hardware specs of their cars. You could maybe try to ask them directly.
What I also just found are their triple front cameras:
https://www.eetasia.com/teslas-hardware-retrofits-for-model-3/
https://www.marklines.com/en/teardown/tesla_model3

xia0long, 2021-09-13:
@Domsall 
Thanks a lot! I tried to set up the cameras with the parameters given by tesla, then I found that someone has already done this, here is the parameter data, hope it can help other people who also have this problem.

```py
CAMERA_CONFIGS = [
    ["front1", carla.Transform(carla.Location(x=1.5, y=-0.1, z=2.4), carla.Rotation(yaw=0)), 45, (0, 0), None],
    ["front2", carla.Transform(carla.Location(x=1.5, y=0, z=2.4), carla.Rotation(yaw=0)), 120, (UNIT_WIDTH, 0), None],
    ["front3", carla.Transform(carla.Location(x=1.5, y=0.1, z=2.4), carla.Rotation(yaw=0)), 90, (2*UNIT_WIDTH, 0), None],
    ["left_front", carla.Transform(carla.Location(x=0, y=-1.1, z=2.4), carla.Rotation(yaw=315)), 90, (0, UNIT_HEIGHT), None],
    ["right_front", carla.Transform(carla.Location(x=0, y=1.1, z=2.4), carla.Rotation(yaw=45)), 90, (2*UNIT_WIDTH, UNIT_HEIGHT), None],
    ["left_rear", carla.Transform(carla.Location(x=1.0, y=-1.1, z=1), carla.Rotation(yaw=225)), 120, (0, 2*UNIT_HEIGHT), None],
    ["rear", carla.Transform(carla.Location(x=-1.5, y=0, z=2.4), carla.Rotation(yaw=180)), 120, (UNIT_WIDTH, 2*UNIT_HEIGHT), None],
    ["right_rear", carla.Transform(carla.Location(x=1.0, y=1.1, z=1), carla.Rotation(yaw=135)), 120, (2*UNIT_WIDTH, 2*UNIT_HEIGHT), None]
]
```

closed 2021-10-04 by xia0long