carla-simulator/carla #4609: Help me plz  the application error (0xc0000142) windows 10

ratchanonth, 2021-09-03:
[image]

NicolaAlbarella, 2021-09-21:
same issue here

joluko, 2021-09-21:
@barasm-hita thanks for your help. I just tried it but unfortunately the same error appears.

barasm-hita, 2021-09-21:
@joluko Your problem must be different, because as you said, version 0.9.11 works fine. So in your case, it may refer to Visual C++ version and/or Windows SDK. I have these versions installed and it works very well:

- Windows 10 SDK v10.0.19041.0
- C++ ATL for latest v142 build tools
- Microsoft Visual C++ 2010
- Microsoft Visual C++ 2013
- Microsoft Visual C++ 2015-2019

I have installed these using Visual Studio 2019 installer and my OS is Windows 10 v10.0.19043

joluko, 2021-09-28:
The error still appears trying to start Carla_0.9.12

@ratchanonth @NicolaAlbarella have you found a solution?

when I open the properties of CarlaUE4.exe the date of the last change is the second of september. was there an update or is there perhaps an error?

wayne-werwolf, 2021-09-28:
I have the same issue. This only occurs with Windows build  [0.9.12](https://github.com/carla-simulator/carla/releases/tag/0.9.12/). The previous release works fine. 

This issue was reproducible on another Windows 10 machine. Both machines meet the hardware requirements.

The requirements mentioned by @barasm-hita are met (although these are not mentioned by the [prerequisites](https://carla.readthedocs.io/en/0.9.12/start_quickstart/) for the pre-built executable).

barasm-hita, 2021-09-28:
@wayne-werwolf See these YouTube videos, maybe they could solve your issue.
https://www.youtube.com/watch?v=byyRbi_gaTs
https://www.youtube.com/watch?v=1YX3bN1YFCM

wayne-werwolf, 2021-09-28:
@barasm-hita Thanks, for the effort 👍. However, these did not solve the issue, and are the usual general troubleshooting advices that come up with a quick google search. I must say, it seems very suspicious that two PCs that are set up very differently, suddenly have the same error while switching from `0.9.11` to `0.9.12`. For me, that points more towards the CARLA distributable than the system environment it is installed on 🤔.

I am trying to pinpoint what might be the cause:
* Did the build system/settings change from `0.9.11` to `0.9.12`, are there any new requirements which are not mentioned? Building it myself is unfortunately not an option.
* It might be due to the Unreal Engine upgrade (which now is `4.26`). I am going to try to install the prerequisites for UE.

Any help or information on this issue is greatly appreciated.

joluko, 2021-09-30:
I also tested it on another Windows 10 PC and its not working either. 

Same error for the nightly build. 

@wayne-werwolf  Have you already tried it with the required UE version?

tayeep, 2021-12-10:
I have the same issue with 0.9.12 and 0.9.13, no luck running it. Version 0.9.11 can open.
[image]

My PC Specs:
- Windows 10 Pro 20H2 build 19042.1348
- Intel Xeon E5-1650 v2
- 32GB RAM
- RTX2060, 497.09
- DirectX 12

bernatx, 2021-12-10:
Hi,
Check the GPU RAM you have. Version 0.9.12 and 0.9.13 of Carla use the new UE4.26 engine version, and also loads the Town10HD_Opt by default, that takes more memory than the previous Town03 that was loaded by default with 0.9.11. Check that is not an issue about how much Gb of GPU. You need probably 6 Gb or more to open Town10HD_Opt.

If that is the case, you can change the default map that will be loaded at start, and set the previous Town03. You need to edit the file **WindowsNoEditor\CarlaUE4\Config\DefaultEngine.ini** and change that map in the section **[/Script/EngineSettings.GameMapsSettings]**. 

You need to replace **=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt** by **=/Game/Carla/Maps/Town03_Opt.Town03_Opt**

Let me know if that fixes the issue.

tayeep, 2021-12-10:
Nope, changing the map doesn't work.
[image]

RTX2060 has 6GB RAM, I guess it should have enough to handle Town10.
[image]

I've already tested 3 machines (2 physicals + 1 virtual machine) and I'm going to run on another 2 more machines with older CPU to see if it works. Will update again.

tayeep, 2021-12-11:
I've tested on two machines, one with Intel+Nvidia, another with AMD. The AMD machines run perfectly even though the display only has 1GB RAM. Is that possible Nvidia driver is the culprit? Is there any settings related to it?

**AMD - Successful run**
[image]

**Intel+Nvidia - failed with same 0xc0000142 error**
- i7-930
- 22GB RAM
- GTX1060
- Win 10 Pro

And then I back to my main machines and tested both Nvidia GAME and STUDIO drivers below with RTX2060, none of it works.
- Studio driver 472.75
- Game driver 497.09

wayne-werwolf, 2021-12-15:
Unfortunately, changing the map as recommended by @bernatx did not fix the issue. Multiple Systems tested, all Intel CPU + Nvidia GPU. As I have no AMD GPU at hand, I cannot test the AMD/Nvidia hypothesis @tayeep.  Current drivers installed.

Tried various flags, e.g. `-quality-level=low`, `-RenderOffScreen`, etc.

I searched for similar issues in other projects that are built with UE4.26, however could not find anything related.

Additionally, I checked for the build requirements to be met on my machine. Curiously, the build docs [image] that `Windows 8.1 SDK` has to be installed. I could not find that in the VS2019 Installer on Windows 10.

Maybe others whose build works could share their specs, and we can pinpoint the cause by looking at the differences?

dinesht27, 2022-05-31:
Hello, 
Was anyone able to resolve this error, and narrow down the solution? Would be really helpful. Seems like it is an accepted gamble issue, whether you machine runs the application or not. No specific direction or suggestion has been stated. 

Should I consider this as a CARLA bug or the Microsoft Error, because the Unreal Engine 4.26 platform executes fine as a standalone, am super confused. 

Any suggestion would do. 

Thanks

OptiAll, 2022-08-10:
Carla does use the older Chrono version 6. Maybe an update to the newest version could help out here. The issue in the chrono repository was not fixed for version 6 after all.
Also Chrono was integrated into the project at version 9.12, which is affected by this issue.

[https://github.com/carla-simulator/carla/blob/master/Util/InstallersWin/install_chrono.bat](https://github.com/carla-simulator/carla/blob/master/Util/InstallersWin/install_chrono.bat)

set CHRONO_VERSION=