carla-simulator/carla #4610: OpticalFlow does not return carla.Image: AttributeError: 'FakeImage' object has no attribute 'save_to_disk'

Paniac, 2021-09-04:
**CARLA version:** 0.9.12

**Platform/OS:** Ubuntu 20.04

**Problem you have experienced:** 
[`OpticalFlowImage.get_color_coded_flow()`](https://carla.readthedocs.io/en/latest/python_api/#getters_10) returns a `FakeImage` object instead of an [`Image`](https://carla.readthedocs.io/en/latest/python_api/#carla.Image) as said in the documentation. `FakeImage` does not have [`Image.save_to_disk()`](https://carla.readthedocs.io/en/latest/python_api/#methods_17) implemented.

**What you expected to happen:** That I can use `save_to_disk()` on the output of `OpticalFlowImage.get_color_coded_flow()`

**Steps to reproduce:** Take a picture with an [OpticalFlowCamera](https://carla.readthedocs.io/en/latest/ref_sensors/#optical-flow-camera). Then call 
```python
data = output.get_color_coded_flow().save_to_disk(path, carla.ColorConverter.Raw)
# `AttributeError: 'FakeImage' object has no attribute 'save_to_disk'`
```

azuzow, 2021-11-10:
Is there any resolution for this issue

jungerm2, 2022-04-12:
Is there any update or workaround for this?

jungerm2, 2022-04-14:
I currently do the following:

```python
import numpy as np
from PIL import Image

image = data.get_color_coded_flow()
buffer = np.frombuffer(image.raw_data, dtype=np.uint8)
buffer = buffer.reshape(image.height, image.width, 4)[..., [2, 1, 0]]  # BGRA -> RGB
Image.fromarray(buffer).save(path)
```

Which works well enough but is probably much slower than `save_to_disk()`.

M-Colley, 2026-10-26:
Draft PR #10029 proposes a disk writer for the color-coded optical-flow result on `ue58-dev`; #10030 is the `ue4-dev` counterpart. They preserve the returned object's existing type and add `save_to_disk` support plus a sensor smoke regression. The UE4 PR currently has merge conflicts. Live optical-flow sensor smoke tests remain unrun.