carla-simulator/carla #4611: Optical flow: wrong flow for moving objects when replay a captured Carla file

MatthiasWildeBusiness, 2021-09-05:
CARLA version: 0.9.12
Platform/OS: Windows and Ubuntu 18.04
Problem you have experienced: Incorrect optical flow, from moving objects are wrong. Looks like only global ego motion is calculated
What you expected to happen: Get correct optical flow data

Steps to reproduce: 
Set a Optical flow camera sensor
Record a Carla file - client.stop_recorder()
Relay the Carla file and store Optical flow camera sensor data  - client.replay_file("record.log"))

knelk, 2022-04-29:
@MatthiasWildeBusiness does this only happen for the playback of a recorded file? What if you record the sensor data during the initial simulation - are the OFs for moving object correct, then?

MatthiasWildeBusiness, 2022-05-09:
Hi,
actually this happens during rendering directly. Replaying a captured file is not supported at all since this is a kind of "set positions" mode and can not calculate flow.