carla-simulator/carla #4612: Adding to PythonAPI - help with boost needed

roque2205, 2021-09-06:
### Discussed in https://github.com/carla-simulator/carla/discussions/4608

<div type='discussions-op-text'>

<sup>Originally posted by **roque2205** September  3, 2021</sup>
Hi there,
I'm trying to add some functionalities to the PythonAPI, but I can't work out how to define API calls with more than one argument in PythonAPI/carla/source/libcarla/Actor.cpp.

Background info: I have created a class that derives from AActor and basically consists of named groups of light elements, which I want to be able to spawn and parameterize using the API. I can spawn, destroy, etc. the class, I also managed to write a function which empties a light group, i.e. deletes all respective light elements.

Extract from Actor.cpp:

      .def("empty_light_group", &cc::LightRig::EmptyGroup, (arg("light_group_name")))  // works fine

I can't quite set up the function which takes in the name of the desired light group and the desired light intensity.

Extract from Actor.cpp:

      .def("set_light_group_intensity", CALL_WITHOUT_GIL_2(cc::LightRig,SetGroupIntensity, std::string, float), (arg("light_group_name"), arg("intensity"))) // evokes a billion of boost errors

      .def("set_light_group_intensity", CONST_CALL_WITHOUT_GIL_2(cc::LightRig,SetGroupIntensity, std::string, float), (arg("light_group_name"), arg("intensity"))) // evokes a billion of boost errors, too

      .def("set_light_group_intensity", &cc::LightRig::SetGroupIntensity, (arg("light_group_name"), arg("intensity"))) // evokes a billion of boost errors as well

For clarity: of course I tried all three variants individually and not at once. ;) I receive C 2977 and C2955 errors, namely ""boost::mpl::list": Too many template arguments" and ""boost::mpl::list": use of class template requires template argument list".

I'd really appreciate help with this.

Best regards
Robert</div>

bernatx, 2021-09-07:
Hi,
We use similar functions like:
`      .def("set_wheel_steer_direction", &cc::Vehicle::SetWheelSteerDirection, (arg("wheel_location")), (arg("angle_in_deg")))`
See that we use each (arg("...")) in its own parethesis. Try to do the same.

roque2205, 2021-09-07:
Thank you so much for your reply, Bernat. I thought I had checked on every possible syntax error, but this particular pair of parentheses had somehow slipped my attention for more than a week of debugging. I'll give it a try!

roque2205, 2021-09-08:
#   define BOOST_MPL_LIMIT_LIST_SIZE 20

THIS is not the maximum number of inputs to a function which is forwarded by boost. This is the maximum length of the variant list of functions which is defined at the end of carla\LibCarla\source\carla\rpc\Command.h. "set_light_group_intensity" was function no. 21.

This cost me weeks. Kill me and make it slow please. :D
By the way,  @bernatx: I guess you'll want to change that parameter to 30 or 40 soon. :D

closed 2021-09-08 by roque2205