carla-simulator/carla #4614: Additional Python API in bre-built carla package？

hungChien, 2021-09-06:
CARLA version: 0.9.12
Platform/OS: ubuntu 18.04
Problem you have experienced:
I am building carla PythonAPI on a ubuntu 18.04 machine with python version default to 3.6. I encountered the following error:
```bash
BuildPythonAPI.sh: Building Python API for Python 3.6.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing dependency_links to source/carla.egg-info/dependency_links.txt
writing top-level names to source/carla.egg-info/top_level.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
clang-8 -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Idependencies/include -I/usr/include/python3.6m -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.6/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.6/source/libcarla/libcarla.o -Ldependencies/lib -o build/lib.linux-x86_64-3.6/carla/libcarla.cpython-36m-x86_64-linux-gnu.so /data/proj/carla/PythonAPI/carla/dependencies/lib/libcarla_client.a /data/proj/carla/PythonAPI/carla/dependencies/lib/librpc.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libboost_filesystem.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libRecast.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libDetour.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libDetourCrowd.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libosm2odr.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libxerces-c.a -lz /data/proj/carla/PythonAPI/carla/dependencies/lib/libproj.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libsqlite3.a /data/proj/carla/PythonAPI/carla/dependencies/lib/libboost_python36.a -lpng -ljpeg -ltiff -lstdc++
/usr/bin/ld: /data/proj/carla/PythonAPI/carla/dependencies/lib/libproj.a(geodesic.c.o): relocation R_X86_64_32S against `.rodata.cst8' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /data/proj/carla/PythonAPI/carla/dependencies/lib/libproj.a(wkt1_generated_parser.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /data/proj/carla/PythonAPI/carla/dependencies/lib/libproj.a(wkt2_generated_parser.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
Util/BuildTools/Linux.mk:87: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1
```
I looked up in the Setup.sh file and found that libproj.a is already built with the `-fPIC` flag, so what is the reason of the error above?
Or alternatively, could you provided more versions of the PythonAPI in the pre-build package(like python 3.6, 3.8, etc)?
I'm sorry if my English would confuse you.

JerryMazeyu, 2022-02-24:
How to solve? I have the same problem.

noobC7, 2022-04-02:
I met exactly the same issue.

gawinwong, 2022-06-06:
Same problem, could someone solve this problem?

gawinwong, 2022-06-06:
I solve the problem by remove anaconda3 and use python 3.6.9.
```
rm -rf Build
make setup
make launch
make PythonAPI
```

ushio2021, 2022-07-20:
Possible Solution:
Replace the build options for PROJ in setup.sh: 
          "-fPIC" --> "-DCMAKE_POSITION_INDEPENDENT_CODE:BOOL=true"

I cannot understand the exact reason (maybe -fPIC is not used?) but I have built the egg file without those exact errors anymore

Blyron, 2026-11-12:
Thanks for the report, and sorry it sat for so long.

**Why the link fails in 0.9.12.** In `Util/BuildTools/Setup.sh`, PROJ is configured with `-DCMAKE_CXX_FLAGS="-std=c++14 -fPIC"`. That flag only applies to C++ sources. The objects the linker rejects (`geodesic.c.o`, `wkt1_generated_parser.c.o`, `wkt2_generated_parser.c.o`) are C files, so they end up in `libproj.a` without `-fPIC`. That is why @ushio2021's change works. To fix it on 0.9.12:

1. In the PROJ `cmake` call in `Setup.sh`, add `-DCMAKE_POSITION_INDEPENDENT_CODE=ON` (or `-DCMAKE_C_FLAGS="-fPIC"`).
2. Delete `Build/proj-install`. `Setup.sh` skips PROJ when that directory exists, so without this the old library is reused.
3. Run `make PythonAPI` again.

You may not need to build at all, though. The 0.9.12 client is on PyPI with wheels for Python 2.7, 3.6, 3.7 and 3.8, so on Ubuntu 18.04 with Python 3.6 `pip install carla==0.9.12` is enough.

**Going forward.** We are migrating to `ue58-dev`, which becomes **CARLA 1.0** (Unreal Engine 5.8). The release packages now ship a prebuilt wheel for **every supported Python version, 3.10 to 3.15**, on both Linux and Windows. They are in `PythonAPI/carla/dist/` inside the package, and the release workflow builds and smoke-tests each of them. Install the one whose `cpXY` tag matches your interpreter, for example:

```sh
pip install CARLA_ROOT/PythonAPI/carla/dist/carla-*-cp312-*.whl
```

so building the Python API yourself to get another Python version is no longer necessary. See the quick start: https://carla-ue5.readthedocs.io/en/latest/start_quickstart/

If you hit problems with the wheels on 1.0 / `ue58-dev`, please open a new issue.

closed 2026-11-12 by Blyron