carla-simulator/carla #4623: Use the Python script in the PythonAPI in the docker environment to access Carla sever outside the docker environment

JackRen88, 2021-09-08:
Hello,carla team!

I start carla server outside a docker environment,and start a Python script(manul_control.py) in Carla’s PythonAPI folder in the docker environment,so I get the message:
`time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000`

How should I establish communication between   the  docker's outside environment and its inside environment?

nick-marnik, 2021-10-06:
To confirm, is the carla server running on the same machine that is running Carla in a Docker container?

SH-Tan, 2021-10-11:
How to use the python in the Carla docker container, when I try to use python, it says there is no python command in this container?

MattRowe18, 2021-11-18:
@JackRen88 can you clarify where you are running your Python scripts? Are you trying to run python from your local machine, or from within the docker container itself? If you are running from your local machine, you need to identify the IP address to reference the Docker container. If you are running scripts inside the container 'localhost' may map directly to the host machine and not to the container itself, depending on your configuration. 

See here for some information about Docker networking https://www.freecodecamp.org/news/how-to-get-a-docker-container-ip-address-explained-with-examples/

closed 2022-04-30 by stale[bot]