carla-simulator/carla #4626: global shader cache file missing

bjavaid1, 2021-09-09:
CARLA version: 0.9.12
Platform/OS: Ubuntu 18.04.5
Problem you have experienced:
When executing launch script (`CarlaUE4.sh`), it gives error that global shader cache file (`GlobalShaderCache-GLSL_150_ES31.bin`) is missing.

The terminal output:
> 4.26.2-0+++UE4+Release-4.26 522 0
> Disabling core dumps.

What you expected to happen: Town launches in spectator view
Steps to reproduce:

1. Download and extract CARLA 0.9.12 release (`CARLA_0.9.12.tar.gz`)
2. Download assets (`AdditionalMaps_0.9.12.tar.gz`) to `Import` directory
3. Run `ImportAssets.sh` script
4. Install client library by making Python virtual environment and running `pip install carla`
5. Run `CarlaUE4.sh`

Other information (documentation you consulted, workarounds you tried):
I am using Quadro RTX 4000 graphics card
The Engine directory contains this file: `GlobalShaderCache-SF_VULKAN_SM5.bin`

FENGShuanglang, 2021-09-11:
try `sudo apt install vulkan-utils`

bjavaid1, 2021-09-15:
> try `sudo apt install vulkan-utils`

That fixed the issue, thanks!

closed 2021-11-18 by MattRowe18