carla-simulator/carla #4628: 'C:\Program' not recognized as command in compiling server or PythonAPI

xilinzhou, 2021-09-10:
CARLA version: 0.9.12
Platform/OS: Windows 10
Problem you have experienced: unable to make PythonAPI or make launch

Hello! We're at the final step in the build (compiling PythonAPI and server). We're running into the error `NMAKE : fatal error U1077: 'C:\Program' : return code '0x1'`. Basically it's not recognizing 'C:\Program' as a command in our preset path to the make bin folder.

Here's the command output:
[image]

xilinzhou, 2021-09-21:
Hi, an update. We’ve downgraded to Carla 0.9.11 and are still encountering the exact same issue. Here’s the other part of the error message.
 
[image]

Axel1092, 2021-11-18:
Hi @xilinzhou, aparently your visual studio installation is broken. Cmake is not able to use it to compile the zlib library.

closed 2022-04-30 by stale[bot]