carla-simulator/carla #4636: how can I run offscreen mode in Windows?

yuunnn, 2021-09-13:
CARLA version:  0.9.12
Platform/OS: Windows 10

Hi, I just built carla following https://carla.readthedocs.io/en/0.9.12/build_windows/.
Now I want to run carla in offscreen mode. How can I do that?

I was confused because I saw in linux mode just run "./CarlaUE4.sh -RenderOffScreen", but there is not bat file in the folder like linux.

I tried 
-cd \Unreal\CarlaUE4\Binaries\Win64
-start CarlaUE4.exe -RenderOffScreen
but get "the global shader cache file is missing" error.

barasm-hita, 2021-09-14:
Hi @yuunnn 
After you have finished building **PythonAPI**, you need to run **_make launch_** command. It will compile libraries and after a while, Unreal Editor app will be launched. Then you can launch Carla server by clicking Launch button from top toolbar.

But if you don't want to launch Editor every time, it's better to use **_make package_** command. It will give you a package that contains binaries for running server without compilation in future. More information have been provided in documentation, the same link you have shared.

At the end, if you do not need to change any codes in Carla server, it's better to use official releases rather than building by yourself.

roque2205, 2021-11-15:
This was so much not what the question was about...
@yuunnn, if you haven't already found the answer to your question, I believe this will help you: https://carla.readthedocs.io/en/0.9.12/adv_rendering_options/#off-screen-mode.

yuunnn, 2021-11-15:
Hi @barasm-hita , thank you！ Yes, I used official releases instead of building on my own, then I can use off-screen mode!

closed 2021-11-18 by MattRowe18

TzabarDolev, 2023-10-04:
Hi, I wrote a blog post about this issue on the CarlaSimBlog, including a code tutorial for off screen rendering.
https://carlasimblog.wordpress.com/2023/10/04/rendering-images-off-screen-in-the-carla-simulator/