carla-simulator/carla #4660: Stuck at "make launch" - make sure you build ShaderCompileWorker.

mimits, 2021-09-19:
CARLA version: 12
Platform/OS: windows
Problem you have experienced: can't launch, last thing I did was make PythonAPI
What you expected to happen: to launch
Steps to reproduce: I did everything as documentation told me
[image]

TrinhTuanHung2021, 2021-09-25:
I also got an error while making launch.  
there are 2 ways to run Carla. If you can't make launch, you can run it directly 

https://github.com/carla-simulator/carla/releases

mimits, 2021-09-25:
Thank you, I did make it, I had a bug in Visual Studio, So I just did everything from the start and this time I waited for visual to build correctly. I got a new problem now, I can't get past "make import" which says that there is no module name "carla"

TrinhTuanHung2021, 2021-09-25:
> Thank you, I did make it, I had a bug in Visual Studio, So I just did everything from the start and this time I waited for visual to build correctly. I got a new problem now, I can't get past "make import" which says that there is no module name "carla"

mimits, 2021-09-25:
[image]
so I install Carla as u say. but not this happens

TrinhTuanHung2021, 2021-09-25:
[image]

mimits, 2021-09-25:
I installed Carla 
[image]

mimits, 2021-09-25:
But then I try to "make import" and the error comes again 
[image]

TrinhTuanHung2021, 2021-09-25:
I see no need to do "make python" or "make launch" commands. After executing the command "pip install carla", run the python files and Carla works

mimits, 2021-09-25:
It is just how it is in documentation, and as I told u :) Carla does work fine, I just cant import any fbx or .xodr files, they way documentation states they should be imported

TrinhTuanHung2021, 2021-09-25:
So I don't know. I also get error when executing "make python" and "make launch" commands.  If you can fix it later, please let me know

mimits, 2021-09-25:
I can enter with "make launch" what error do u have?

TrinhTuanHung2021, 2021-09-25:
Do you use facebook or something to make communication easier?

mimits, 2021-09-25:
Andrej#7032 discord?

TrinhTuanHung2021, 2021-09-25:
> Andrej#7032 discord?

is this a nickname or what? I searched but couldn't find it

mimits, 2021-09-25:
I have sent u a request on FB

MattRowe18, 2021-11-18:
@bernatx can you take a look at this

closed 2022-04-29 by stale[bot]