carla-simulator/carla #4666: Loading xodr file gives strange output

JoyMazumder, 2021-09-20:
Hello everyone, I am experimenting with xodr file and tried to load it into carla. I run the following command `python3 config.py -x opendrive/TownBig.xodr` and got the output something like this. The load are not aligned properly

[image]

I followed this tutorial to install Carla https://carla.readthedocs.io/en/latest/start_quickstart/#a-debian-carla-installation (Debian installation). OS linux 18.04, UE `4.26.2-0+++UE4+Release-4.26`

JoyMazumder, 2021-09-20:
Running `manual_control.py` works fine
[image]

closed 2021-09-20 by JoyMazumder

barasm-hita, 2021-09-25:
After loading xodr map into CARLA, you can move camera via holding mouse click and WASD keys on keyboard.
You see roads as invisible because the camera is viewing them from bottom then just move camera upward and roads will show up correctly.

CaffreyXu, 2022-04-13:
hello, I have the same question as you. The lane line can't be seen. Do you solve the question? Thank you so much

CaffreyXu, 2022-04-13:
@JoyMazumder

hitabm, 2022-04-16:
> hello, I have the same question as you. The lane line can't be seen. Do you solve the question? Thank you so much

Hi, you can draw string on each lane. It will show red signs for each lane in server side window. Use this code:
`way_points = world.get_map().generate_waypoints(distance=10)
for i, wp in enumerate(way_points):
        world.debug.draw_string(wp.transform.location, 'O', draw_shadow=False,
                                color=carla.Color(r=255, g=0, b=0), life_time=1200.0,
                                persistent_lines=True)`

CaffreyXu, 2022-04-18:
@hitabm  Thank you so much! I saw many useful function in "debug" library, I will try to draw the line i want.