carla-simulator/carla #4668: Use custom maps in package build after importing in source build

sashajenner, 2021-09-21:
CARLA version: 0.9.12-dirty
Platform/OS: Ubuntu 18.04.6 LTS

I have built CARLA from source and created custom maps using Roadrunner. These were successfully imported into the build. I can start Unreal Editor using `make launch`, open these maps and successfully run the Carla server on them.

### Problem
However, I am hoping to run the Carla server on these custom maps without using Unreal Editor and without a display at all.

### What I have tried
I understand the packaged version supports this and that I can download it or just use `make package` to create it. However, I can't seem to be able to load these maps in a Carla server launched from a package build.

I have tried to put the custom map data into the `CarlaUE4/Content/map_package` directory of the package build and after launching the Carla server and running `python3 config.py -m map_package` or using the Python API `client.load_world(map_package)` I get the following error from the server side:
```
LowLevelFatalError [File:Unknown] [Line: 149]
A FRenderResource was deleted without being released first!
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
I believe `map_package` is being found because a different error occurs when a random map name is given in which case the server does not crash. However, I'm not sure how to proceed, if this is possible to achieve and if there is a simpler solution to my problem.

chengguizi, 2021-11-08:
Hi @sashajenner , I believe I have a similar issue on 0.9.12

For me the setup is a little different. But I have similar symptoms:

- I can run things perfectly with `make launch` and play in UE4 editor mode
- This works both in Epic quality and Low quality

However, After I `make package ARGS=...`, the pacakged carla will crash if i turn on low quality mode AND if i spawn camera sensors.

Something like below
```
DRI_PRIME=1 ~/carla/CarlaUE4.sh -norelativemousemode -quality-level=Low
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
INTEL-MESA: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

WARNING: cannot parse georeference: ''. Using default values. 
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

If I run without cameras, it is normal
```
DRI_PRIME=1 ~/carla/CarlaUE4.sh -norelativemousemode -quality-level=Low
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
INTEL-MESA: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

WARNING: cannot parse georeference: ''. Using default values.
```

One thing to note that, the low quality works fine with all other sensors, including semantic lidars.

@sashajenner  are you using low quality mode, or just the default quality.

closed 2022-04-29 by stale[bot]

muety, 2024-02-20:
Got the same issue here :wave: