carla-simulator/carla #4680: Missing waypoints and spawn points in xodr maps

barasm-hita, 2021-09-25:
CARLA version: 0.9.12
Platform/OS: Windows 10 10.0.19043

I used exporting **OSM** road and conversion of it to **xodr** exactly how it has been explained in docs. All cases that I have been trying worked without any problem. But, as I need to simulate raceway roads (like [this ](https://www.openstreetmap.org/export#map=16/52.0293/11.2805)one) it causes an error. The error isn't in conversion, all things happen very fine and without any bugs. But when I try to run some examples like automatic_control.py and manual_control.py, it says that the **world** has no **spawn points**, and it cannot create waypoints.

closed 2022-05-02 by stale[bot]

ar4development, 2022-06-10:
@hitabm did you find any solution for that?

hitabm, 2022-06-24:
@ar4development sorry for late response. I didn't find any principled way, but now I can use these sort of osm files and convert them to xodr in my own way!
As you know osm files are editable with any text editor, so I open them with my editor (like VS code) and find tags related to these racing roads. Then change their type to any of types that carla osm converter does support (i.e. highway). For more examples of my explanation check my own project repo at [here](https://github.com/hitabm/carla-vehicle-testing) and its map folder.