carla-simulator/carla #4681: Maneuver Behavior

iammix, 2021-09-25:
CARLA version: 0.9.12
Platform/OS: Windows 10

I would like to perform a maneuver using one of the default agents (Basic, Behavior). To be more specific I would like to perform an overtake. Is it possible or I can achieve that only from autopilot(using the traffic control manager)

barasm-hita, 2021-09-25:
I have a situation similar to your one. So, I created a new example of vehicle controlling based on existing automatic_control.py example. In my example, I describe new behavior type that I call it **_Custom_**. May be you will need to do something similar to achieve your own purpose. [Here ](https://github.com/barasm-hita/carla/blob/vehicle_behavior_testing/PythonAPI/examples/control_vehicle.py)is my example.

iammix, 2021-09-29:
@barasm-hita  Thank you a lot for your response. However, in a previous commit, there was a method in the behavior agent class that was performing an overtake. I cannot understand why it was removed. Do you have any clue about it?

barasm-hita, 2021-09-30:
@iammix You're welcome. Not really, unfortunately I have no idea about the removal. Its better to ask the main development team, but maybe they have other plans to recover it with better performance in later versions.

iammix, 2021-10-06:
> @iammix You're welcome. Not really, unfortunately I have no idea about the removal. Its better to ask the main development team, but maybe they have other plans to recover it with better performance in later versions.

Thank you @barasm-hita

closed 2021-10-06 by iammix

arjunsk07, 2023-03-06:
@hitabm Can you provide the example code for a reference. Because it is not available now.

hitabm, 2023-04-13:
> @hitabm Can you provide the example code for a reference. Because it is not available now.

Sure, sorry for the corrupted link. I did a fix on it you can try it now. Same [here](https://github.com/hitabm/carla-vehicle-testing/blob/main/examples/control_vehicle.py).