carla-simulator/carla #4687: Cannot fina a compatible Vulkan Driver (ICD)?

YunJin-Zhang, 2021-09-28:
**Carla Version: latest**
**Platform: Carla Docker in Ubuntu 18.04**

After pulling and running the latest CARLA docker:
`docker run --privileged --gpus all -it --net=host -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlarism/carla:latest /bin/bash`

then I tried to run ./CarlaUE.sh as a test in it, a window popped up:
[image]
It looks like there isn't a Vulkan driver in the docker.

and something appeared in the shell:
[image]
$ ./CarlaUE4.sh
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found

What should I do? Thanks!

mateocervilla, 2021-09-28:
Check if your DISPLAY variable is defined:
`echo $DISPLAY`
If not, set it with:
`export DISPLAY=:0`

YunJin-Zhang, 2021-09-29:
> Check if your DISPLAY variable is defined: `echo $DISPLAY` If not, set it with: `export DISPLAY=:0`
> 
> You can also try to add -vulkan parameter and variable -e SDL_VIDEODRIVER=x11: `docker run --privileged --gpus all --net=host -e DISPLAY=$DISPLAY -e SDL_VIDEODRIVER=x11 -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlasim/carla:latest /bin/bash ./CarlaUE4.sh -vulkan`

Thanks for your answer! It's not the reason you pointed out, but thank you all the same.
Why the UE.sh cannot find a compatible driver is for I missed any parameters when I installed the NVIDIA Graphics driver. The most usual choice for parameters in installation of NVIDIA Graphics driver is:
`sudo bash NVIDIA-Linux-x86_64-430.26.run –-no-opengl-files –-no-x-check`

so actually the OpenGL or Vulkan wasn't installed alongside.
Then I re-installed the NVIDIA driver with ignoring the two alternative parameters, everything is fine now.
Thank you very much! Best wishes.

closed 2021-09-29 by YunJin-Zhang