carla-simulator/carla #4689: Python API wheels for Linux and Windows for Python 3.9 on PyPi?

VonAncken, 2021-09-28:
CARLA version: 0.9.12
Platform/OS: Windows 10 and Ubuntu 20.04 LTS

Currently there are wheels for Python 2.7, 3.6, 3.7 and 3.8 available on PyPi for the CARLA Python API. Is it possible to provide wheels for Windows and Linux for Python 3.9 as well?
As far as I can tell, the CARLA Python API is now being built in Amazon AWS. Could you extend the build for Python 3.9?
I know I could build the wheels myself, but I think I'm not the only one who needs them.

joel-mb, 2021-11-18:
Hi @VonAncken,
We are planning to provide wheels for Windows and Ubuntu for Python 3.9 but it is not straightforward right now. We've detected some incompatibilities with the current version of boost we are using and Python 3.9. So probably we need to upgrade it. I'll leave this issue open to report any progress on this topic.

dk42, 2022-01-20:
Hi @joel-mb,

could you elaborate on the incompatibilities? I managed to compile the CARLA Python API for Python 3.9. And it seems to work. Am I missing something?

gdippolito, 2022-06-06:
I think @joel-mb is referring to the official Python wheel of carla available in pypi (https://pypi.org/project/carla/) not supporting Python 3.9

gdippolito, 2022-07-25:
@bernatx would it be possible to provide a wheels for Python 3.9 whenever a new (0.9.14) release will be cut? 

At the moment there is no wheel available in Pypi (https://pypi.org/project/carla/) which support Python 3.9.

Sultan91, 2023-08-12:
Still no support for python 3.9?

xavisolesoft, 2024-04-26:
Thank you for your feedback!
We created a ticked to build python API for more python versions. Will fix this soon.

https://github.com/carla-simulator/carla/issues/7473

closed 2024-04-26 by xavisolesoft