carla-simulator/carla #4690: Building Carla on Windows Error

tripleaamin, 2021-09-28:


closed 2021-09-28 by tripleaamin

atharva25061997, 2021-11-09:
I have the same error. Have you found a solution to this? If yes, please help me.

tripleaamin, 2021-11-09:
> I have the same error. Have you found a solution to this? If yes, please help me.

I believe I had to upgrade my version of cmake. Altho I ran into more issues installing from source.

atharva25061997, 2021-11-09:
I have the latest version of Cmake. I am still getting the error. Do you have any idea what can be done?

physihan, 2021-12-30:
We maybe need new version of Visual Studio, My current version is Visual Studio 2017.

lakshjaisinghani, 2022-01-17:
Use Visual Studio 2019, it works for me!

LBroKerR, 2023-10-29:
i use cmake 3.9.0 and visual studio 2019 and i still got the problem. i tried a lot of cmake version above the 3.9.0 one, but i got error for those
could u help me pls?

datj, 2024-10-12:
> @atharva25061997 Use Visual Studio 2019, it works for me!

Good day everybody here!

This topic still needs some resurrecting especially for those having problems still & those confused

Could someone please give tips or links how to acquire/build Unreal for ARM Windows, Unreal being a Carla prerequisite?