carla-simulator/carla #4697: Carla crashes if collision happens and collision sensor attached to vehicle.

JoyMazumder, 2021-09-30:
Hello everyone,

I am using Carla 0.9.12 Debian build in ubuntu 18.04 and faced an issue. If I attach a collision sensor to a vehicle and if any collision happens then it crashes. If no collision sensor is attached to the vehicle it works fine. The problem arises after attaching the collision sensor. I have provided my code below
```
actor_list = []

class CollisionSensor(object):
    def __init__(self, parent_actor):
        self.sensor = None
        self.history = []
        self._parent = parent_actor
        world = self._parent.get_world()
        bp = world.get_blueprint_library().find('sensor.other.collision')
        self.sensor = world.spawn_actor(bp, carla.Transform(), attach_to=self._parent)

        weak_self = weakref.ref(self)
        self.sensor.listen(lambda event: CollisionSensor._on_collision(weak_self, event))

    def get_collision_history(self):
        history = collections.defaultdict(int)
        for frame, intensity in self.history:
            history[frame] += intensity
        return history

    def destroy(self):
        self.sensor.destroy()

    @staticmethod
    def _on_collision(weak_self, event):
        self = weak_self()
        if not self:
            return
 
        impulse = event.normal_impulse
        intensity = math.sqrt(impulse.x**2 + impulse.y**2 + impulse.z**2)
        self.history.append((event.frame, intensity))
        if len(self.history) > 4000:
            self.history.pop(0)

def main():
    client = carla.Client('localhost', 2000)
    client.set_timeout(2.0)
    try:
        world = client.get_world()
        client.start_recorder('/media/joy/Dataset/Carla_experiment/PythonAPI/examples/saved/recording01.log',True)

        blueprint_library = world.get_blueprint_library()
        bp = blueprint_library.filter('model3')[0]
        bp.set_attribute('role_name', 'hero')
        spawn_point = random.choice(world.get_map().get_spawn_points())
        vehicle = world.spawn_actor(bp, spawn_point)
        vehicle.apply_control(carla.VehicleControl(throttle=1.0, steer=0.0))
        actor_list.append(vehicle)

        # do something with this sensor
        collision_sensor = CollisionSensor(vehicle)

        spectator = world.get_spectator()
        world_snapshot = world.wait_for_tick()
        spectator.set_transform(vehicle.get_transform())

        while True:
            world_snapshot = world.wait_for_tick()

    finally:
        client.stop_recorder()
        print('destroying actors')
        for actor in actor_list:
            actor.destroy()

 ```
Any suggestion will be appreciated. Thank you

LucasFKobernic, 2021-10-05:
I tried your code on my computer (windows 10, CARLA 0.9.12 built from source) and it works without crashing

closed 2022-05-02 by stale[bot]

TigerStone93, 2022-11-02:
> @TigerStone93 I tried on windows (build from source) as @LucasFKobernic suggested and It worked.

Then does it mean that the same code shows different result according to its OS?

That's strange... it might be a bug. Really annoying.

Thank you for your fast response!

JoyMaz222, 2022-11-02:
yes. same code provided different result.