carla-simulator/carla #4698: Can't fully work on the selected GPU on a multi-GPU environment

suniique, 2021-09-30:
The CARLA 0.9.12 can't fully work on the GPU selected by UE4's argument `-graphicsadapter=N`. Specifically, when set N >= 1, 
- The main threads of CARLA do run on GPU #N, 
- But there are also some very small usages (around 1-4 MB) on the first physical GPU (GPU #0). 

```
Thu Sep 30 21:30:35 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.73.01    Driver Version: 460.73.01    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX TIT...  On   | 00000000:05:00.0 Off |                  N/A |
| 22%   37C    P8    15W / 250W |      3MiB / 12212MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  GeForce GTX TIT...  On   | 00000000:09:00.0 Off |                  N/A |
| 67%   85C    P0   155W / 250W |   4058MiB / 12212MiB |     99%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A     25116      G   ...x/CarlaUE4-Linux-Shipping        1MiB |
|    1   N/A  N/A     25116    C+G   ...x/CarlaUE4-Linux-Shipping     4052MiB |
+-----------------------------------------------------------------------------+
```

Setting the environment variable `CUDA_VISIBLE_DEVICES=1` will not resolve this problem. In the previous versions, when using OpenGL's backend, there's a solution of setting the SDL's environment variables, but these will not apply the 0.9.12 since it only works on Vulkan.

It seems that CARLA will not respect the CUDA's environment variable. This will cause some problems in a shared GPU cluster, where the GPUs are logically ordered by these variables. Do you have any clues about this issue?

p.s. This issue is not about running CARLA with multiple GPUs, but about running it with only one specified GPU (not affected other GPUs).

--
CARLA version: 0.9.12
Platform/OS: Debian

suniique, 2021-10-01:
I listed the threads that rely on GPU#0 by `lsof -eT | grep nvidia0`. It seems that a lot of threads (`TaskGraphNP, TaskGraphBP, TaskGraphHP, StatsThread, PoolThread 1 ~ 9, OnlineA-ance, HttpManager, RHIThread, FAsyncPurge, RenderThread, RTHeartBeat, CarlaUE4-Linux-Shipping, ...`) will use this GPU. The list is like this: 

```
COMMAND           PID   TID TASKCMD              FD        TYPE   DEVICE  SIZE/OFF     NODE  NAME
CarlaUE4-Linux- 25116 25117 TaskGraphNP          mem       CHR    195,0                22188 /dev/nvidia0
CarlaUE4-Linux- 25116 25117 TaskGraphNP          6u        CHR    195,0       0t0      22188 /dev/nvidia0
...
CarlaUE4-Linux- 25116 25117 TaskGraphNP          24u       CHR    195,0       0t0      22188 /dev/nvidia0
...
```

Based on the threads' names, I think not every thread will use the GPU function. Therefore, I guess these usages may just come from the device tests?

closed 2022-05-02 by stale[bot]

zzj403, 2023-10-13:
not solved in 0.9.14, sad

AOOOOOA, 2023-12-02:
still struggle with this problem on 0.9.13.