carla-simulator/carla #4701: Setting Auto-Pilot variable speed and variable position inside the lane for each non-player-vehicle for increasing realism

lucalazzaroni, 2021-10-01:
CARLA version: 0.9.11
Platform/OS: Windows 10

Hi everyone. I am working with the Carla simulator for generating Autonomous Driving scenarios (follow, cut-in, etc.). Since some of the scenarios I created require traffic, I need non-player-vehicles to behave as randomly as possible in order to increase realism. In particular, I need them to have variable speeds changing over time but also to move left and right inside a lane with a little offset if possible. I tried to play with the TrafficManager but with no significant results. Can you please suggest to me how could I solve this kind of problem?
Thanks in advance for your help.

YuffieHuang, 2021-10-13:
CARLA version: 0.9.11
Platform/OS: Ubuntu 18.04

Hi @glopezdiest. I'm having a similar issue. I want to make 20% of vehicles to be aggressive (speeding and tailgating) and 20% of vehicles to be cautious (drive slowly and keep long distance to front vehicle). I followed https://carla.readthedocs.io/en/0.9.11/adv_traffic_manager/#running-multiple-traffic-managers and added these lines to spawn_npc.py. I don't know what "my_vehicles" is in the document.

    traffic_manager.global_percentage_speed_difference(100.0)  # drives at speed limit
    traffic_manager.set_global_distance_to_leading_vehicle(2.0)  # keeps normal following distance

    aggressive_car = []  # a list for aggressive vehicles
    cautious_car = []   # a list for cautious vehicles

    for i in vehicles_list:   # The vehicles_list contains all vehicle IDs (a list of integers).
         traffic_manager.set_auto_lane_change(i, True)   # allow lane change
         percent = random.rand()
         if percent <= 0.2:  # select 20% cars to be aggressive
             aggressive_car.append(i)
         elif percent >= 0.8:    # select 20% cars to be cautious
             cautious_car.append(i)
        
    # define aggressive behaviors
    # traffic_manager.ignore_lights_percentage(aggressive_car, 100)  # ignore traffic lights
    traffic_manager.distance_to_leading_vehicle(aggressive_car, 0.0)    # tailgating
    traffic_manager.vehicle_percentage_speed_difference(aggressive_car, -20.0)  # 20% over speed limit
    # define cautious behaviors
    traffic_manager.distance_to_leading_vehicle(cautious_car, 5.0)  # long following distance
    traffic_manager.vehicle_percentage_speed_difference(cautious_car, 80.0)  # 20% under speed limit

However, I have encountered several errors. For example:
1.  traffic_manager.set_auto_lane_change(i, True)
AttributeError: 'TrafficManager' object has no attribute 'set_auto_lane_change'
AttributeError: 'int' object has no attribute 'set_autopilot'

2. traffic_manager.distance_to_leading_vehicle(danger_car, 0.0)
Boost.Python.ArgumentError: Python argument types in TrafficManager.distance_to_leading_vehicle(TrafficManager, list, float)
did not match C++ signature: distance_to_leading_vehicle(carla::traffic_manager::TrafficManager {lvalue}, boost::shared_ptr<carla::client::Actor>, float)

Can you please help me out? Thank you so much!

glopezdiest, 2021-10-13:
I think you are mixing several concepts, as you have several minor issues which might be pilling up.

First, the TM has two different types of parameters, the global ones, where all vehicles are affected, and the individual ones, where a specific actor is affected. You can use several TMs, one for each type of behavior you want, and then set their global parameters:
```python
aggresive_tm = client.get_trafficmanager(4000)
aggresive_tm.global_percentage_speed_difference(-20.0)
aggresive_tm.set_global_distance_to_leading_vehicle(0.0)

cautious_tm = client.get_trafficmanager(4001)
cautious_tm.global_percentage_speed_difference(20.0) 
cautious_tm.set_global_distance_to_leading_vehicle(5.0)

normal_tm = client.get_trafficmanager(4002)

# Plus add the vehicles to the TM you want using 'vehicle.set_autopilot(tm_port)'
```
or use the individual parameters to decide their behavior
2) Create one TM and use individual parameters for all vehicles.
```python
tm = client.get_trafficmanager()
for vehicle in vehicles_list:
     percent = random.rand()
     vehicle.set_autopilot()
     if percent <= 0.2:  # select 20% cars to be aggressive
         tm.vehicle_percentage_speed_difference(vehicle, -20.0)
         tm.distance_to_leading_vehicle(vehicle, 0.0)
     elif percent >= 0.8:    # select 20% cars to be cautious
         tm.vehicle_percentage_speed_difference(vehicle, 20.0)
         tm.distance_to_leading_vehicle(vehicle, 5.0)
```

About your other issues:
> traffic_manager.set_auto_lane_change(i, True)

The parameter is *traffic_manager.auto_lane_change()*, and it needs an actor instance, not an actor ID

> traffic_manager.distance_to_leading_vehicle(aggressive_car, 0.0)

This is an individual parameters, so it needs an actor as input, not a list

lucalazzaroni, 2021-10-18:
Sorry to re-open the issue, but I faced another problem related to this one. I managed to use the traffic manager as you @glopezdiest suggested, now I can make vehicles going at different speeds, thank you. I would also like to manage the distance between vehicles, but I noticed that if it is lower than 3 meters, vehicles do not follow their rules anymore. For instance, if I make a "Follow leading vehicle scenario" and the hero car distance from the next one is lower than 3 meters, it does not follow the other car.

TavernaBrunoMarco, 2021-10-18:
Hi @glopezdiest, i try to explain the problem: the problem is the distance that i want set. in a traffic scenario i want to set the distance between the vheicles and i want that the distance is = 1 . If i write dist_vehicles = 3 or less the distance between the vheicles is not 3 but is the default distance. Instead if i decide to set dist_vehicles = 4 or more the vheicles maintaining the distance that i set. The code of the photo is in atomic_behaviors.py in the line 1505.
[image]
I have loaded 2 videos in which in the first the dist_vehicles is 10 and it is possible to see that the vehicles trying to maintain a distance = 10 but in the second, in which the dist_vehicles is 1, the vehicles brakes before reaching 1 meter of distance from the vheicle in front.

[Traffic_dist=10.avi.zip](https://github.com/carla-simulator/carla/files/7365643/Traffic_dist.10.avi.zip)
[Traffic_dist=1.avi.zip](https://github.com/carla-simulator/carla/files/7365645/Traffic_dist.1.avi.zip)

Furthermore i have the same problem of the distance for the scenario in which the adversary cut the road to hero. If i want that the adversary starting to cut the road to hero when the adversary have a distance = 3 or less from hero, the adversary don't cut the road but if the distance is = 4 or more the adversary cut the road.
[image]
I have loaded also 2 videos of the cut road scenario. 

[Yes_CutIn.avi.zip](https://github.com/carla-simulator/carla/files/7365330/Yes_CutIn.avi.zip)
[No_CutIn.avi.zip](https://github.com/carla-simulator/carla/files/7365334/No_CutIn.avi.zip)

So i don't understand why everywhere i setting distance = 4 or more the things works  but if i setting distance = 3 or less no.  
Thanks in advance for your help.

closed 2021-12-01 by MattRowe18