carla-simulator/carla #4703: Cannot 'make launch' when building carla 0.9.12 fails with error 6

Liuzy0908, 2021-10-01:


Liuzy0908, 2021-10-01:
CARLA version:0.9.12
Platform/OS:WIN10
------------------------------------------------------------------------------------------------------
Description:

 D:\Program Files\Epic Games\UE_4.26\Engine\Source\Runtime\CoreUObject\Public\Templates/Casts.h(442): error C4541: "dynamic_cast" is used for the polymorphic type "FActorData" with /GR-; Can lead to unpredictable behavior
  F:\0-Carla_0.9.12\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Actor/CarlaActor.h(169): note: View a reference to the instantiation of the function template being compiled, "FVehicleData *UE4Casts_Private::DynamicCast<T*,ObjectType>(From *)"
          with
          [
              T=FVehicleData,
              ObjectType=FActorData,
              From=FActorData
          ]
  F:/0-Carla_0.9.12/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/CarlaActor.cpp(635): note: View of the compiler is a function template instantiation "T * FCarlaActor: : GetActorData < FVehicleData > (void)"
          with
          [
              T=FVehicleData
          ]
make: *** [CarlaUE4Editor] error 6

---------------------------------------------------------------------------------------------------------------------------

Previous work was done by installing the instruction manual. And the UE4_ROOT already points to UE4.26.  How can I solve this problem - [CarlaUE4Editor] error 6?

RobTranquillo, 2021-12-06:
I'am also got the "error 6" on Windows 10 on `make lauch`

I stated following on an already closed issue, a couple of days ago: https://github.com/carla-simulator/carla/issues/2776#issuecomment-985382318. Maybe it helps to find out something:

I think my problem arises here in [line 130 of BuildCarlaUE4.bat](https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/Util/BuildTools/BuildCarlaUE4.bat#L130)
Because an added output of `echo "%ROOT_PATH%Unreal/CarlaUE4/CarlaUE4.uproject"` shows that slashes are in use instead of backslashes `"c:/Carla/carla/Unreal/CarlaUE4/CarlaUE4.uproject"`. If I change this manually, 
the process continues until the next error occurs: `make: *** [CarlaUE4Editor] Fehler 6` ("Fehler" means error in German).

my working paths for clarification:

d:\CarlaUnrealEngine\  => CarlaUnreal repo with successfully completed binaries build
c:\Carla\carla   =>  Carla repo, which produces the above build error
c:\Program Files\Epic Games\UE_4.26  =>  installed UE4

I'am on branch https://github.com/carla-simulator/carla/tree/master
and https://github.com/CarlaUnreal/UnrealEngine/tree/carla

And the last output before the error appears while using self compiled CarlaUnreal Version 4.26. (set via UE4_ROOT variable):

```
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(29): warning C4458: declaration of 'Character' hides class member
  D:\CarlaUnrealEngine\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(55): warning C4458: declaration of 'Character' hides class member
  D:\CarlaUnrealEngine\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(90): warning C4458: declaration of 'Character' hides class member
  D:\CarlaUnrealEngine\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(131): warning C4458: declaration of 'Character' hides class member
  D:\CarlaUnrealEngine\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(151): warning C4458: declaration of 'Character' hides class member
  D:\CarlaUnrealEngine\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(171): warning C4458: declaration of 'Character' hides class member
  D:\CarlaUnrealEngine\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  D:\CarlaUnrealEngine\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectBaseUtility.h(546): warning C4996: 'UVehicleAnimInstance': PhysX is deprecated. Use the UVehicleAnimationInstance from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\CarlaUnrealEngine\Engine\Source\Runtime\CoreUObject\Public\Templates/Casts.h(143): note: see reference to function template instantiation 'bool UObjectBaseUtility::IsA<To>(void) const' being compiled
          with
          [
              To=UVehicleAnimInstance
          ]
  D:\CarlaUnrealEngine\Engine\Source\Runtime\CoreUObject\Public\Templates/Casts.h(142): note: while compiling class template member function 'To *TCastImpl<From,To,ECastType::UObjectToUObject>::DoCast(UObject *)'
          with
          [
              To=UVehicleAnimInstance,
              From=UAnimInstance
          ]
  D:\CarlaUnrealEngine\Engine\Source\Runtime\CoreUObject\Public\Templates/Casts.h(219): note: see reference to function template instantiation 'To *TCastImpl<From,To,ECastType::UObjectToUObject>::DoCast(UObject *)' being compiled
          with
          [
              To=UVehicleAnimInstance,
              From=UAnimInstance
          ]
  D:\CarlaUnrealEngine\Engine\Source\Runtime\CoreUObject\Public\Templates/Casts.h(219): note: see reference to class template instantiation 'TCastImpl<From,To,ECastType::UObjectToUObject>' being compiled
          with
          [
              From=UAnimInstance,
              To=UVehicleAnimInstance
          ]
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(572): note: see reference to function template instantiation 'To *Cast<UVehicleAnimInstance,UAnimInstance>(From *)' being compiled
          with
          [
              To=UVehicleAnimInstance,
              From=UAnimInstance
          ]
  D:\CarlaUnrealEngine\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectBaseUtility.h(546): warning C4996: 'UVehicleAnimInstance::StaticClass': PhysX is deprecated. Use the UVehicleAnimationInstance from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
[... cut ...]
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(476): warning C4458: Deklaration von "Role" blendet Klassenmember aus
  C:\Program Files\Epic Games\UE_4.26\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(541): note: Siehe Deklaration von "AActor::Role"
  C:/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(491): warning C4458: Deklaration von "Role" blendet Klassenmember aus
  C:\Program Files\Epic Games\UE_4.26\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(541): note: Siehe Deklaration von "AActor::Role"
make: *** [CarlaUE4Editor] Fehler 6
```

RobTranquillo, 2022-02-01:
OK. I uninstalled the standard Unreal version. Reset and updated Carla and CarlaUnrealEngine repositories. Followed the build instructions (https://carla.readthedocs.io/en/latest/build_windows/) exactly (it is important to run `GenerateProjectFiles.bat` every time something has changed. Checked the path of UE4_ROOT, it must point to the CarlaUnrealEngine repository directory (the documentation is missing the correct folder, because it only talks about the installed Unreal version). And then, fortunately, the build runs fine.

RamyaManjunath19, 2022-09-13:
> 

Hi @RobTranquillo I face the same error. I have set the path variable of as shown in the below image. I still get the same error. I also followed the carla document link that you have shared. Can you please help me resolve this error ? Thank you.

[image]

ballonJourn, 2023-12-20:
> > 
> 
> Hi @RobTranquillo I face the same error. I have set the path variable of as shown in the below image. I still get the same error. I also followed the carla document link that you have shared. Can you please help me resolve this error ? Thank you.
> 
> [image]

maybe u should change the path to C:\Program Files\Epic Games\UE_4.26\