carla-simulator/carla #4711: Carla 0.9.12 crashes with scenario runner

semmelknoedel, 2021-10-06:
Hi,

I have recently upgraded to Carla 0.9.12, however it seems to crash all the time I start the scenario_runner as well. It is independent and happens with any scenario.

I get the following console output:

Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

I use Ubuntu 20.04 LTS with
NVIDIA GeForce RTX 2080 SUPER
and driver version 470.

I use the precompiled release version, but also the docker container crashes upon start.
Before that I didn't have this problem, however, I always used the -opengl flag, since for my system vulkan always had problems before (older vresions and ubuntu 18.04).

Do you need any further information?

nick-marnik, 2021-10-06:
The opengl flag was removed in Carla 0.9.12, which is based on UE4 4.26. I suggest making sure you have the latest NVIDIA drivers installed and try running the following command to validate the Vulkan API is working: `vulkaninfo | less`

semmelknoedel, 2021-10-07:
`vulkaninfo | less` returns a lot of output. Is there anything in particular I have to look at? I also tried carla 0.9.11 with Vulkan yesterday and it was running without any problems also with the scenario_runner.
Also the Nvidia driver version 470 is the newest I can find.

nick-marnik, 2021-10-08:
Interesting - it sounds like the GPU drivers and the Vulkan API are working as expected. When you tried Carla 0.9.11, are you using the opengl flag?

It may be worth trying a source build of Carla 0.9.12, but I'm guessing the error will be similar. What is the exact error you get when trying to run the Carla 0.9.12 Docker image?

semmelknoedel, 2021-10-13:
I actually tried both variants with Carla 0.9.11 with and without opengl flag. Both are working fine. (I used the precompiled release version). Also, 0.9.12 is only crashing when using the scenario_runner. Without the scenario_runner it is working.

When I start the docker container of Carla 0.9.12, I get the following output:
`sudo docker run --privileged --gpus all --net=host -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlasim/carla:0.9.12 /bin/bash ./CarlaUE4.sh -RenderOffScreen
/home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: error while loading shared libraries: libomp.so.5: cannot open shared object file: No such file or directory` 
The same for the non-off screen version.
The docker for 0.9.11, however, is running fine.

I might be able to try start Carla 0.9.12 from Unreal, since I am in the office on Monday. Then I will give you a short update if this is working.

semmelknoedel, 2021-11-01:
@nick-marnik : thanks for the link. The docker container of carla 0.9.12 is now running and working with the scenario_runner.

I also updated to Nvidia driver to 495. However, the precompiled version is still not working. Since that works for me-> ticket closed.

closed 2021-11-01 by semmelknoedel

abviv, 2021-11-19:
> @nick-marnik : thanks for the link. The docker container of carla 0.9.12 is now running and working with the scenario_runner.
> 
> I also updated to Nvidia driver to 495. However, the precompiled version is still not working. Since that works for me-> ticket closed.

I am pretty new to CARLA and i was wondering do you have some sort of docs about how you're able to run scenario_runner with docker?

semmelknoedel, 2021-11-19:
> I am pretty new to CARLA and i was wondering do you have some sort of docs about how you're able to run scenario_runner with docker?

Hi @abviv : if you don't use any special configurations (e.g. different ports) with your docker container, it should work the same way as with any other Carla version (pre-compiled release or running in Unreal). How to start a scenario is pretty well documented here: [scenario_runner docu](https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_scenariorunner.md)

Just start the docker container and then the scenario runner as explained in the link above in scetion B. Of cousre, the versions of the scenario runner and the docker container have to match as well.