carla-simulator/carla #4713: Texture Mask Missing in RoadMaster - My beautiful road wear lines are gone!!

jimmyw404, 2021-10-06:
CARLA version: 0.9.12

Platform/OS: Ubuntu / Windows
Problem you have experienced: Road wear lines are missing
What you expected to happen: Road wear lines to be present
Steps to reproduce: Load the Town10HD map from either the package or built from source, compare to prior versions.
Other information (documentation you consulted, workarounds you tried):

The RoadMaster-based Material Instances in the 0.9.12 version of CARLA are the Scalar Parameter Values / Texture Parameter Values that enable blending a Texture Mask to add road wear lines to the map are missing. This feature is documented in https://carla.readthedocs.io/en/0.9.12/tuto_M_custom_road_painter/#paint-the-road. The effect can readily be seeng by comparing Town10HD in 0.9.12 to prior versions.

The below two images show the material instance + Town10HD map in CARLA 0.9.10, which shows off the road wear lines
[image]
[image]

The below two images show the material instance + Town10HD map in CARLA 0.9.12, which does not have road wear lines.
[image]
[image]

XGodina, 2021-10-14:
Hi @jimmyw404,

We are now working on finishing a maintenance release, which will soon come out. Where this problem will be fixed. If you need to fix it you just have to select the material of the road and put the mask of the Town10. 
[image]

Sorry for the inconvenience it may have caused.

jimmyw404, 2021-10-14:
> Hi @jimmyw404,
> 
> We are now working on finishing a maintenance release, which will soon come out. Where this problem will be fixed. If you need to fix it you just have to select the material of the road and put the mask of the Town10. [image]
> 
> Sorry for the inconvenience it may have caused.

Works great, thanks!!!!

Closing this ticket...

closed 2021-10-14 by jimmyw404