carla-simulator/carla #4716: cannot run Carla using nvidia GPU under Linux despite PRIME set to nvidia

julienlau, 2021-10-07:
This issue is linked to closed issues :
#3920
#3490
#3196
#2396
#2490

I don't think this should be played down, because it seems most Linux users using dual GPU have this issue and either quit trying either switch to using the intel GPU instead of their available Nvidia.

CARLA version: carla-simulator-0.9.12 4.26.2-0+++UE4+Release-4.26 522 0
Platform/OS: Ubuntu 20.04.3 LTS
Problem you have experienced: cannot launch carla

Steps to reproduce:
I think Carla is grabbing the first active GPU device (message about `MESA` points toward intel GPU), instead of the one selected by PRIME. See screenshot below.

[image]

Using `-opengl` do not make things work, I get a message box and then the same error + coredump as without `-opengl`

[image]

- How to run Carla unde Linux when using Prime selector and not blacklisting devices ?
- How to check which GPU is using Carla ?

rrrushan, 2021-10-09:
I am facing the same issue at the moment.
If a solution is found, please comment.
I will do the same.

chinmay-d, 2021-10-12:
Same issue faced by me too :
```
$ ./CarlaUE4.sh -quality-level=Epic

4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x340003b
  Serial number of failed request:  285
  Current serial number in output stream:  295
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)
```

Let me know if anyone finds a solution.. thanks

julienlau, 2021-10-14:
I also did the test with carla 0.9.11, but I have exactly the same problem.
Are you sure you did not switch to running on your intel GPU ?
With carla version 0.9.11 the run option `-opengl` enables me to launch carla over the intel GPU, but it is very very slow and this is not what I want.
With carla version 0.9.12 this option does not work (even to run it over the intel GPU).

chinmay-d, 2021-10-14:
I also get same error when I try to run ./CarlaUE4.sh ...... also -opengl dosen't work for me. 
This is the error I get.
````
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x440003b
  Serial number of failed request:  286
  Current serial number in output stream:  296
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)
````

smk32, 2021-10-21:
Just was having the same issue. Looks like a workaround is to force Carla to use nvida by: 
`DRI_PRIME=1 ./CarlaUE4.sh`
This allowed me to run the app successfully.

caillotantoine, 2021-10-21:
I experience the same issue. I used the use the `-opengl` option to run the previous version. There is also an issue to run Docker file #4755 . Thus, currently, I have no way using Carla 0.9.12.

th3superuser, 2021-11-02:
unfortunately, I still have the same error using `./CarlaUE4.sh -prefernvidia` and also `-opengl` doesn't work for me either, anyone have an idea? I can provide trace/logs if that's helpful.

closed 2022-04-30 by stale[bot]

lehoangan2906, 2024-03-13:
> I experience the same issue. I used the use the `-opengl` option to run the previous version. There is also an issue to run Docker file #4755 . I found a workaround adding the option `-prefernvidia` which seemed to solve the problem.
> 
> Full command line : `./CarlaUE4.sh -prefernvidia`

Really appreciate this, it worked out for me.