carla-simulator/carla #4755: XDG_RUNTIME_DIR not set for Carla 0.9.12 in docker container

itsPauV, 2021-10-15:
Hi, 

I was trying to setup a carla server in a docker container with a display (the spectator mode).
Which works well with the following docker command and carla version 0.9.10.1:
`docker run  -p 2000-2002:2000-2002  --cpuset-cpus="0-5"  --runtime=nvidia  --gpus 'all,"capabilities=graphics,utility,display,video,compute"'  -e DISPLAY=$DISPLAY  -v /tmp/.X11-unix:/tmp/.X11-unix -it --privileged carlasim/carla:0.9.10.1 ./CarlaUE4.sh -opengl $1`

But if I try to execute that same command with carla version 0.9.12 I get the following error message:

> 4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.

I also get the same error message when I try to start it with the command specified in the docs:
`docker run -e DISPLAY=$DISPLAY --net=host --gpus all --runtime=nvidia carlasim/carla:<version> /bin/bash CarlaUE4.sh -opengl <-additonal-carla-flags>`

I can manage to start the 0.9.12 server to start without a display but I'd like to get it to work with a display.

Thank you in advance.

CARLA version: 0.9.12
Platform/OS: Ubuntu 20.04 / Docker

gouchaoer, 2021-10-18:
@Axel1092  @mawi42 @jackbart94

any suggestions?

Cenbylin, 2021-11-11:
Any update for this issue ?

CCHAVES-AVS, 2022-06-17:
I also encountered the same problem.

Running on WSL2 (ubuntu 20.04), nvidia driver 512.15

Any update on this issue?

```
sudo docker run --privileged --gpus all --net=host -e DISPLAY=$DISPLAY carlasim/carla:0.9.13 /bin/bash ./CarlaUE4.sh
sh: 1: xdg-user-dir: not found
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
```

nahsimai, 2022-07-15:
> I see the same error when following the CARLA in Docker instructions. I am running 0.9.13 in Docker on a fresh install of Ubuntu 20.04.
> 
> The following command works for me, but it requires the `xhost +` command which may not be desirable for security reasons. Either way, I hope this a useful data point. Use at your own risk.
> 
> First, allow connections from any host.
> 
> ```
> xhost +
> 
> access control disabled, clients can connect from any host
> ```
> 
> Next, start the simulator.
> 
> ```
> docker run --name carlaserver \
>     -e SDL_VIDEODRIVER=x11 \
>     -e DISPLAY=$DISPLAY \
>     -e XAUTHORITY=$XAUTHORITY \
>     -v /tmp/.X11-unix:/tmp/.X11-unix \
>     -v $XAUTHORITY:$XAUTHORITY \
>     -it --gpus 'all,"capabilities=graphics,utility,display,video,compute"' \
>     -p 2000-2002:2000-2002 carlasim/carla:0.9.13 \
>     ./CarlaUE4.sh -quality-level=Epic
> 
>     4.26.2-0+++UE4+Release-4.26 522 0
>     Disabling core dumps.
>     sh: 1: xdg-user-dir: not found
>     ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
>    ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
>    (SOUND CARD ERRORS CONTINUE...)
>    
> ```
> 
> The simulator opens and seems to work fine, but I have not done any further testing.

As mentioned [here](https://github.com/carla-simulator/carla/issues/3431#issuecomment-708786234), you can also use `xhost local:root` instead of `xhost +`. It works for me at least.

muety, 2023-08-15:
Got the same issue here. Any solutions for the `carlasim/carla` image?

mohkan1, 2024-12-30:
> As mentioned [here](https://github.com/carla-simulator/carla/issues/3431#issuecomment-708786234), you can also use `xhost local:root` instead of `xhost +`. It works for me at least.

This command works for me also

mostafatouny, 2025-03-04:
> I see the same error when following the CARLA in Docker instructions. I am running 0.9.13 in Docker on a fresh install of Ubuntu 20.04.
> The following command works for me, but it requires the `xhost [...]

Confirmed to work