carla-simulator/carla #4775: Out of video memory trying to allocate a rendering resource (CARLA 0.9.12)

JunadMQ, 2021-10-21:
CARLA version: 0.9.12
Platform/OS: Windows 10
GPU: NVIDIA Geforce GTX 1650 (8GB)
RAM: 8GB

Problem: Running "CarlaUE.exe" gives an error message **"Out of video memory trying to allocate rendering resource.** Make sure your video card has the minimum required memory....." and exits

Other information (documentation you consulted, workarounds you tried):

*Version 0.9.11 and earlier seems to work fine
*Drivers updated to latest version
*Running Carla with the _"-quality-level=Low", "--no-rendering", "-RenderOffScreen"_ flags also yield the same error
*Increasing Virtual Memory (15000 MB) did not help
*Installed latest version of C++ Runtimes (https://docs.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist?view=msvc-160) - was prompted to do this in the initial run of "CarlaUE.exe"

Yosefnoorani, 2022-10-03:
Thank!

Very helpfull!

excep, 2022-11-08:
I had the same problem and this solved it. Thank you 
But what does -dx11 do exactly !!? just trying to understand what's wrong

Farghal1, 2023-04-01:
Hello,
I did write the command in the cmd , but this error appears 'CarlaUE4' is not recognized as an internal or external command,
operable program or batch file.

zaakki-ahamed, 2023-04-01:
> I had the same problem and this solved it. Thank you But what does -dx11 do exactly !!? just trying to understand what's wrong

From what I gather, it runs Carla in DirectX 11 mode. Some machines have compatibility issues running on DirectX 12. I'm not aware of the root cause for this issue.

zaakki-ahamed, 2023-04-01:
> Hello, I did write the command in the cmd , but this error appears 'CarlaUE4' is not recognized as an internal or external command, operable program or batch file.

Are you sure you are running the command from the correct folder? In Command Prompt, you must navigate to the folder containing the "CarlaUE4.exe" executable. Try running it once you do that.

Farghal1, 2023-04-01:
yes, i navigate to the location where Carlaue4.exe exist and then run the command

zaakki-ahamed, 2023-04-01:
> yes, i navigate to the location where Carlaue4.exe exist and then run the command

May I know the CARLA version you are using? A screenshot of the folder and the error might also help

Farghal1, 2023-04-01:
Im using Carla 0.9.14
and this is the path "D:\Materials\Bachelor's Material\CARLA\CARLA_0.9.14 (1)\WindowsNoEditor"
and this is the [error] 'CarlaUE4' is not recognized as an internal or external command,
operable program or batch file.

Farghal1, 2023-04-01:
I solved the problem and Carla now works perfectly.
I needed to add the path of CarlaUE4.exe to the environment variables, open the command prompt and navigate to its location, then run it using this command "CarlaUE4 -dx11"

Samrat-WhitehatJR896, 2024-06-16:
This solved it. Thanks :)