carla-simulator/carla #4826: Over 2x Drop in Rendering Performance in 0.9.12

jimmyw404, 2021-11-02:
CARLA version: 0.9.12
Platform/OS: Ubuntu 20.04.2 LTS
Problem you have experienced: benchmark performance is significantly lower in 0.9.12 vs previous releases
What you expected to happen: benchmarking performance to be similar between 0.9.12 vs previous releases
Steps to reproduce: Run PythonAPI/util/performance_benchmark.py on different versions with the same sensor configurations.
Other information (documentation you consulted, workarounds you tried):

During normal (non-benchmarked) operation I empirically noticed a significant performance drop on multiple machines when switching to CARLA 0.9.12. I downloaded packages for the latest CARLA (dev commit 85da613f as of Nov 1, 2021), CARLA 0.9.12 and all the previous versions and ran the performance_benchmark.py utility on them.

Comparing the results in Town03 and 4x 1920x1080 camera I get the below performance:

[image]

I configured the sensor options with identical parameters:
```
sensors03_from912 = [{'type': 'sensor.camera.rgb', 'x': 0.7, 'y': 0.0, 'z': 1.60, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
              'width': 1920, 'height': 1080, 'fov': 100, 'label': 'sensors03_from912. cam-1920x1080'},
              {'type': 'sensor.camera.rgb', 'x': 0.7, 'y': 0.0, 'z': 1.60, 'roll': 0.0, 'pitch': 0.0, 'yaw': 90.0,
              'width': 1920, 'height': 1080, 'fov': 100, 'label': 'cam-1920x1080'},
              {'type': 'sensor.camera.rgb', 'x': 0.7, 'y': 0.
0, 'z': 1.60, 'roll': 0.0, 'pitch': 0.0, 'yaw': 180.0,
              'width': 1920, 'height': 1080, 'fov': 100, 'label': 'cam-1920x1080'},
              {'type': 'sensor.camera.rgb', 'x': 0.7, 'y': 0.0, 'z': 1.60, 'roll': 0.0, 'pitch': 0.0, 'yaw': 270.0,
              'width': 1920, 'height': 1080, 'fov': 100, 'label': 'cam-1920x1080'}]
```

I used ClearNoon as the weather preset
```weather00 = { 'parameter' : carla.WeatherParameters.ClearNoon, 'name' : 'ClearNoon'}```

I downloaded the packages from these tags: https://github.com/carla-simulator/carla/tags and https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/Dev/CARLA_Latest.tar.gz

I'm using an NVIDIA RTX 2060 with driver version 470.63.01 on an i7-8700 with 32GB of RAM on Ubuntu 20.
```
j@j-desktop:~/carla/pkg/CARLA_0.9.12/PythonAPI/util$ nvidia-smi 
Mon Nov  1 16:51:51 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.63.01    Driver Version: 470.63.01    CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| 57%   51C    P3    34W / 160W |   3010MiB /  5931MiB |      9%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A       873      G   /usr/lib/xorg/Xorg                120MiB |
|    0   N/A  N/A      1615      G   /usr/lib/xorg/Xorg                347MiB |
|    0   N/A  N/A      1750      G   /usr/bin/gnome-shell               70MiB |
|    0   N/A  N/A      2189      G   ...AAAAAAAAA= --shared-files       28MiB |
|    0   N/A  N/A      5093      G   ...AAAAAAAAA= --shared-files       69MiB |
|    0   N/A  N/A     10312    C+G   ...x/CarlaUE4-Linux-Shipping     2350MiB |
|    0   N/A  N/A     10480      G   python3                             2MiB |
+-----------------------------------------------------------------------------+

j@j-desktop:~/carla/pkg/CARLA_0.9.12/PythonAPI/util$ sudo lshw -C display | grep product
       product: TU106 [GeForce RTX 2060 Rev. A]

j@j-desktop:~/carla/pkg/CARLA_0.9.12/PythonAPI/util$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"

j@j-desktop:~/carla/pkg/CARLA_0.9.12/PythonAPI/util$ lscpu | grep "Model name"
Model name:                      Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz

j@j-desktop:~/carla/pkg/CARLA_0.9.12/PythonAPI/util$ cat /proc/meminfo 
MemTotal:       32811036 kB
```

Benchmark output:
[CARLA_0.9.8_Benchmark.md](https://github.com/carla-simulator/carla/files/7460341/CARLA_0.9.8_Benchmark.md)
[CARLA_0.9.9.4_Benchmark.md](https://github.com/carla-simulator/carla/files/7460342/CARLA_0.9.9.4_Benchmark.md)
[CARLA_0.9.10_Benchmark.md](https://github.com/carla-simulator/carla/files/7460343/CARLA_0.9.10_Benchmark.md)
[CARLA_0.9.10.1_Benchmark.md](https://github.com/carla-simulator/carla/files/7455995/CARLA_0.9.10.1_Benchmark.md)
[CARLA_0.9.11_Benchmark.md](https://github.com/carla-simulator/carla/files/7455996/CARLA_0.9.11_Benchmark.md)
[CARLA_0.9.12_Benchmark.md](https://github.com/carla-simulator/carla/files/7455997/CARLA_0.9.12_Benchmark.md)
[CARLA_0.9.12_Benchmark_Run2.md](https://github.com/carla-simulator/carla/files/7460344/CARLA_0.9.12_Benchmark_Run2.md)
[CARLA_0.9.12_Latest_85da613f_Benchmark.md](https://github.com/carla-simulator/carla/files/7456088/CARLA_0.9.12_Latest_85da613f_Benchmark.md)

**Is there anything I'm doing wrong to get this performance decrease??? Is there something deficient with my system I can fix? I've noticed this on multiple systems but they have similar configurations. Let me know if I can try anything to help improve this, I really like 0.9.12's improvements but can't upgrade with this performance hit!!!**

jimmyw404, 2021-11-12:
Issue appears related to 

https://github.com/carla-simulator/carla/issues/3836

Scaling up cameras, especially on a PC with a powerful GPU (Testing against an RTX 3080 Ti) does not increase GPU Utilization.

jimmyw404, 2021-11-12:
Also related to

https://github.com/carla-simulator/carla/issues/3891

xqrdot, 2021-11-24:
Can confirm a **30%** drop for Camera performance between `0.9.11` and `0.9.13`

Identical sensor configurations, Town01, 1000 Samples.

[image]

Setup information:
Ubuntu 18.04, RTX 2070 Max-Q, i7-8750H CPU @ 2.20GHz × 12
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.82.00    Driver Version: 470.82.00    CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   58C    P8    10W /  N/A |    537MiB /  7982MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1391      G   /usr/lib/xorg/Xorg                 28MiB |
|    0   N/A  N/A      1571      G   /usr/bin/gnome-shell               69MiB |
|    0   N/A  N/A      2132      G   /usr/lib/xorg/Xorg                179MiB |
|    0   N/A  N/A      2316      G   /usr/bin/gnome-shell               40MiB |
|    0   N/A  N/A     14434      G   ...AAAAAAAAA= --shared-files       65MiB |
|    0   N/A  N/A     15488      G   ...AAAAAAAAA= --shared-files      145MiB |
+-----------------------------------------------------------------------------+
```

closed 2022-05-02 by stale[bot]