carla-simulator/carla #4828: Accessing individual wheel torque and rpm

everestmax, 2021-11-02:
Hi, 

I would like to be able to access each wheels torque and RPM on the vehicle, but cant work out the best way to do so. 

When I enable telemetry on the manual_control.py script, I can see that there are many graphs of tyre forces displayed in the server window, and it is these I would like to ideally access through the client. 

I am running Unreal Engine 4.26.2 and Python 3.6.

Any help would be greatly appreciated.

glopezdiest, 2021-12-01:
Hey @everestmax, all the current exposed information about the wheels is in the [carla.WheelPhysicsControl](https://carla.readthedocs.io/en/latest/python_api/#carlawheelphysicscontrol), which you can access doing `actor.get_physics_control().wheels`. However, I don't think data like the wheels RPM is exposed.

closed 2021-12-01 by glopezdiest

AOOOOOA, 2023-03-24:
> Hey @everestmax, all the current exposed information about the wheels is in the [carla.WheelPhysicsControl](https://carla.readthedocs.io/en/latest/python_api/#carlawheelphysicscontrol), which you [...]

Hi, is it exposed now? Can we directly access the related info of wheels? Thanks！

MurilloGuindani, 2024-07-18:
> Hi, is it exposed now? Can we directly access the related info of wheels? Thanks！

Yes, was recently merged into the dev branch.
[Expose Telemetry Data #5153 ](https://github.com/carla-simulator/carla/pull/5153)

BBArikL, 2024-08-26:
I just added a PR to fix the access to the "wheels" attribute of VehicleTelemetryData : #8078