carla-simulator/carla #4834: about docker off-screen mode

gengtsh, 2021-11-04:
hi, 
i try to run carla in a server (running with ubuntu 18.04, without a display, nvidia Quadro P6000) from local windows 10 MobaXterm.
i install docker and nvidia container toolkit following instruction. 
when i run : 
sudo docker run --privileged --gpus all --net=host -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlasim/carla:0.9.12 /bin/bash ./CarlaUE4.sh -RenderOffScreen
output: sh: 1:xdg-user-dir: not found 
but, when i run docker ps, i found image carlasim/carla:0.9.12 command "/bin/bash ./CarlaUE4.sh" are created 10 minutes ago.
there is nothing happened except i got one line output sh: 1:xdg-user-dir: not found 

is it successful or not? i suppose there is something wrong, but don't know how to fix it.
can someone help me out and provide a simple tutorial (example) to run carla with docker on off-screen mode?

LetStarFly, 2021-12-29:
do you slovle the problem？ I want to do same thing as you. It's too diffcult.

Kin-Zhang, 2022-01-17:
I test about the carla 0.9.10.1 from docker hub with docker pull, and I find that with display the carla can run successfully
``bash
docker run -it -e DISPLAY=$DISPLAY --net=host --gpus all --name kin_carla2 carlasim/carla:0.9.10.1 /bin/bash
./CarlaUE4.sh -opengl
```
but When I tried to without display, it just exit and didn't run the carla without any error.
![image](https://user-images.githubusercontent.com/35365764/149772777-962abe23-071d-4785-b83d-75ca3ab40cf9.png)

Kin-Zhang, 2022-01-17:
> I test about the carla 0.9.10.1 from docker hub with docker pull, and I find that with display the carla can run successfully
> 
> ```shell
> docker run -it -e DISPLAY=$DISPLAY --net=host --gpus all --name kin_carla2 carlasim/carla:0.9.10.1 /bin/bash
> ./CarlaUE4.sh -opengl
> ```
> 
> but When I tried to without display, it just exit and didn't run the carla without any error. Like here, I will try the 0.9.12 to see if it can run without display which is off-screen mode
> 
> [image]

OK, I just test on Carla 0.9.12 which won't have this problem.

closed 2023-06-02 by stale[bot]