carla-simulator/carla #4835: make rebuild fails because of missing directories

thbake, 2021-11-04:
Hi Carla Team!

I am having some trouble with the command ```make rebuild```. After following your guide on how to create sensors one is supposed to rebuild the project, but I get the following error: 
```
x86_64-linux-gnu-gcc: error: /home/bake/carla/PythonAPI/carla/dependencies/lib/libxerces-c.a: No such file or directory
x86_64-linux-gnu-gcc: error: /home/bake/carla/PythonAPI/carla/dependencies/lib/libproj.a: No such file or directory
x86_64-linux-gnu-gcc: error: /home/bake/carla/PythonAPI/carla/dependencies/lib/libsqlite3.a: No such file or directory
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
make: *** [Util/BuildTools/Linux.mk:38: rebuild] Error 1
```
I don't know why this is the case since I installed all the dependencies and I also was able to build and launch Carla successfully. Do you have any input on this?

Thanks in advance.

CARLA version: 0.9.12
Platform/OS: Ubuntu 20.04

Cenbylin, 2021-12-02:
I have the same issue. The system is Ubuntu 20 as well.

MaikolDrechsler, 2021-12-07:
I am facing the same issue. Also in Ubuntu 20.04. and Carla Version 0.9..12. 
Did you find any solution? 

Thank you in advance

thbake, 2021-12-07:
Hey guys! No, I haven't I am first trying to integrate LibCarla in a different way and waiting for a response of the developers. If someone replies or I find I way I'll let you know

IKAROS93, 2022-01-19:
the solution is:
**DO NOT** `make rebuild`
just `make clean` and `make PythonAPI` and `make launch`

ucbmpc2021, 2022-02-26:
Same issue here! Have you found a solution?

MaikolDrechsler, 2022-02-28:
@ucbmpc2021 
I am just using make launch. If you really need to do a rebuild, you can manually copy the three libraries. 

In this case: 

-  copy the three libraries from CARLA_FOLDER/Build/proj-install/
-  `make rebuild`
- Wait until the CARLA_FOLDER/PythonAPI/carla/dependencies/lib/ folder is created and paste the libraries inside. 
- Wait until the rebuild process is done. 

It is just a workaround, but it worked for me. 
OBS: In the same way, you will need to `make launch` to initiate the simulator

ucbmpc2021, 2022-03-26:
@MaikolDrechsler Thanks for your tips! Very helpful.

Bonifatius94, 2022-07-11:
I'm trying to build the code inside Docker, so *make launch* isn't an option for me.
Can someone please fix the *make rebuild* command? :wink: