carla-simulator/carla #4839: Can we set initial velocity via the traffic manager?

wujingda, 2021-11-05:
Hi there,

Can we set the initial velocity of the vehicle actors under the control of the traffic manager?
I checked the document and found the function "tm.set_target_velocity()", but it still makes the actors speed up from zero. 
For my designed scenario, an initial velocity is needed since the scenario is small and cannot afford the speed-up process.

Thanks in advance.

glopezdiest, 2021-11-18:
Hey @wujingda I think you are mixing concepts. The Traffic Manager cannot change the initial speed of the vehicles, but can still use the `actor.set_target_velocity()` on an actor controlled by the TM, and it'll work just fine. make sure the speed vector is set to move the vehicle to the front:
```py
yaw = actor.get_transform().rotation.yaw * (math.pi / 180)
vx = init_speed * math.cos(yaw)
vy = init_speed * math.sin(yaw)
self._actor.set_target_velocity(carla.Vector3D(vx, vy, 0))
``` 

With this, you are most set for 99% of use cases, but as you said *an initial velocity is needed since the scenario is small and cannot afford the speed-up process.*, and I've recently worked on this, here is some in-depth review you might want to take into account.

Firstly, you have to take into account that this just forces the speed of the velocity to that value. This doesn't start the engine of the vehicle nor does it make it change gears to the correct value, so for the next frames the vehicle will most likely reduce speed, until the engine has properly started.

You could solve this by constantly applying the target velocity, but that will cause collisions to be very unrealistic (on collision, physics make the vehicle stop, but next frame you are forcing its speed back up, so it collides again at full speed). 

Ideally, you'd want to constantly apply the target speed and add a [collision sensor](https://carla.readthedocs.io/en/latest/ref_sensors/#collision-detector) that, on collision, stops the setting of the velocity. This is what I've been done [here](https://github.com/carla-simulator/carla/blob/glopez/constant_velocity_agent/PythonAPI/carla/agents/navigation/constant_velocity_agent.py).

wujingda, 2021-11-19:
Hi @glopezdiest, Thanks for the warm help! I will try your method.

closed 2021-11-24 by glopezdiest