carla-simulator/carla #4844: Is there another option to download the content?

jitpinun, 2021-11-06:
I'm currently trying to compile the latest Carla (0.9.12) in my PC. However, during the content downloading stage, it somehow is very slow (will take more than 600 days to finish the download). Is there any other way to download the content.

FYI, downloading in Singapore.
----

$ ./Update.sh
--2021-11-07 01:47:25--  http://carla-assets.s3.amazonaws.com/20210730_564bcdc.tar.gz
Resolving carla-assets.s3.amazonaws.com (carla-assets.s3.amazonaws.com)... 52.95.154.75
Connecting to carla-assets.s3.amazonaws.com (carla-assets.s3.amazonaws.com)|52.95.154.75|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14466455072 (13G) [application/x-tar]
Saving to: ‘Content.tar.gz’
Content.tar.gz        0%[                   ] 344.58K   481 B/s    eta 616d

MattRowe18, 2021-11-25:
@jitpinun please try the following alternative links to find the packages for your version of CARLA https://github.com/carla-simulator/carla/blob/master/Docs/download.md

If you still have problems please reopen this issue

closed 2021-11-25 by MattRowe18

zf-account, 2024-03-01:
you can visit https://github.com/carla-simulator/carla/releases，Find the corresponding version，AdditionalMaps_0.9.14.tar.gz is the content.