carla-simulator/carla #4859: make PythonAPI error

atharva25061997, 2021-11-09:
C:\Carla\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Carla\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Carla\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Carla\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Carla\carla\Build\" --zlib-install-dir "C:\Carla\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Carla\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Carla\carla\Build\"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called "rpclib-src".
    -[install_rpclib]: Generating build...
CMake Error: The source directory "C:/Carla/carla/Build/rpclib-src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_rpclib]: [CMAKE ERROR] An error ocurred while executing the cmake.
    -[install_rpclib]: [CMAKE ERROR] Possible causes:
    -[install_rpclib]:                - Make sure "CMake" is installed.
    -[install_rpclib]:                - Make sure it is available on your Windows "path".
    -[install_rpclib]:                - CMake 3.9.0 or higher is required.
    -[install_rpclib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Fehler 1

Can anybody help me with this problem?

nick-marnik, 2021-11-18:
What version of Carla? Per the instructions, cmake needs to be installed manually: https://cmake.org/download/

atharva25061997, 2021-11-19:
I have cmake installed on my system. The version too is the required one. Yet this error persists.

nick-marnik, 2021-11-19:
A command like `where cmake` will only work in a `x64 Native Tools Command Prompt for VS 2019` window. The VS version depends on the Carla release:
0.9.11 or older: VS 2017
0.9.12 or newer: VS 2019

[image]

atharva25061997, 2021-11-22:
I have already tried this. I even tried both versions but it is not working.

atharva25061997, 2021-11-23:
> Can you share some screenshots and output of what you have tried so far? It sounds silly, but a reboot may be helpful because Windows can be finicky about how environment variables are refreshed inside of command prompts.

I have tried everything that has been suggested in the documentation. Even rebooting didn't help. When i will try again in coming days and the problem persists then I will share the screenshots.

MiaoMiaoMeng, 2021-11-25:
try to install those lib manually
which you can find in /carla/Util/InstallersWin/ 
and the url and folder‘s name is in those .bat file
@atharva25061997

Ku-Buqi, 2022-01-07:
F:\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  12
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "F:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "F:\carla\Build\"
    -[install_zlib]: Extracting zlib from "zlib-1.2.11.zip".
    -[install_zlib]: Removing "zlib-1.2.11.zip"
系统找不到指定的文件。
    -[install_zlib]: Creating "F:\carla\Build\zlib-source\build"
CMake Error: The source directory "F:/carla/Build/zlib-source" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 错误 1

I had a similar problem.Can anyone help me?

JEFFWU017, 2022-02-24:
I had a similar problem.Can anyone help me?

ritabt, 2022-04-04:
The problem is that zlib version 1.2.11 doesn't have a website anymore so when running make the error means that it can't download the zipped zlib. 
I was able to fix it by going to carla/InstallersWin/install_zlib.bat and changing line 51 to "set ZLIB_VERSION=1.2.12" instead of "set ZLIB_VERSION=1.2.11"

JEFFWU017, 2022-04-04:
Yeah, I got it. I will try it firstly. Anyway, thank you very much.

luckysouthchou, 2022-04-30:
lif

> The problem is that zlib version 1.2.11 doesn't have a website anymore so when running make the error means that it can't download the zipped zlib. I was able to fix it by going to carla/Util/InstallersWin/install_zlib.bat and changing line 51 to "set ZLIB_VERSION=1.2.12" instead of "set ZLIB_VERSION=1.2.11" Then delete the build folder in carla/ and run "make PythonAPi" again

life saver, then everything should be ok!!!!

longshujing, 2022-05-15:
Hanging ladder

JohnxLiang, 2022-05-15:
> 問題是 zlib 版本 1.2.11 沒有網站了，所以運行時 make 錯誤意味著它無法下載壓縮的 zlib。 我能夠通過轉到 carla/Util/InstallersWin/install_zlib.bat 並將第 51 行更改為“set ZLIB_VERSION=1.2.12”而不是“set ZLIB_VERSION=1.2.11”來修復它 然後刪除 carla/ 中的構建文件夾和再次運行“make PythonAPi”

Thanks.

JohnxLiang, 2022-05-15:
> The problem is that zlib version 1.2.11 doesn't have a website anymore so when running make the error means that it can't download the zipped zlib. I was able to fix it by going to [...]

Thanks. It can work.

Nozoomhs, 2022-06-10:
> The problem is that zlib version 1.2.11 doesn't have a website anymore so when running make the error means that it can't download the zipped zlib. I was able to fix it by going to [...]

I have the same issue yet this fix doesn't seem to work for me. any other ideas?

l1th1um, 2022-08-21:
I counter this issue, and solved it with delete the Build directory

dangli1989, 2022-10-03:
> The problem is that zlib version 1.2.11 doesn't have a website anymore so when running make the error means that it can't download the zipped zlib. I was able to fix it by going to [...]

It works for me

vemi-lab, 2022-11-03:
The current zlib version is 1.2.13 so, if you encounter this problem make sure to change line 51 in carla/Util/InstallersWin/install_zlib.bat to "set ZLIB_VERSION=1.2.13", then delete the Build folder in the Carla Root folder and try again.

tonmoay, 2023-01-16:
Try installing Git and add it to the PATH. Hope this helps!

tyotili, 2023-03-02:
Adding this to PATH and then deleting the build folder and restarting everything worked for me:

C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer\Git\

liujiawei725, 2023-05-30:
Try to download the zlib1213.zip file from https://www.zlib.net/, and rename the zip file as zlib-1.2.13.zip.  Place the renamed the zip file in the carla/build directory and execute make PythonAPI

Canio95, 2023-10-19:
If you face this problem still, I fixed it by changing line 51 in the carla/Util/InstallersWin/install_zlib.bat to set ZLIB_VERSION=1.3

CStubenvoll, 2024-02-16:
zlib is now on version 1.3.1 and changing line 51 in the aforementioned fix to that solved it for me.

AlwaysLELE, 2024-06-30:
我使用这个更改版本号，并删除build文件夹的方法还是构建失败

Emilyhelin, 2024-07-05:
> I have tried everything that has been suggested in the documentation. Even rebooting didn't help. When i will try again in coming days and the problem persists then I will share the screenshots.

I solved it by substituting make PythonAPI for make PythonAPI GENERATOR="Visual Studio 17 2022".

Nuaagggyk, 2024-07-18:
D:\CARLA_0.9.11\carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "D:\CARLA_0.9.11\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "D:\CARLA_0.9.11\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\CARLA_0.9.11\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "D:\CARLA_0.9.11\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
系统找不到指定的路径。
已复制         0 个文件。
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "D:\CARLA_0.9.11\carla\Build\" --zlib-install-dir "D:\CARLA_0.9.11\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "D:\CARLA_0.9.11\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:\CARLA_0.9.11\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called "rpclib-src".
    -[install_rpclib]: Generating build...
CMake Warning:
  Ignoring extra path from command line:

   ".."

CMake Error: The source directory "D:/CARLA_0.9.11/carla/Build/rpclib-src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_rpclib]: [CMAKE ERROR] An error ocurred while executing the cmake.
    -[install_rpclib]: [CMAKE ERROR] Possible causes:
    -[install_rpclib]:                - Make sure "CMake" is installed.
    -[install_rpclib]:                - Make sure it is available on your Windows "path".
    -[install_rpclib]:                - CMake 3.9.0 or higher is required.
    -[install_rpclib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 错误 1

I have already  changed  line 51 to "set ZLIB_VERSION=1.3.1" instead of "set ZLIB_VERSION=1.2.11" , but it doesn't work for me

Emilyhelin, 2024-07-28:
> @Emilyhelin ，Are you using vs22 for compiling

Yes. But I found that it will generate an error afterwards. I recommend using vs 2019.
Thanks to reminding from seominseok on discord, we can download vs2019 from this link: https://reddit.com/r/VisualStudio/comments/q4dm0u/cant_download_visual_studio_community_19/?rdt=59046

royranma, 2024-07-30:
C:\Users\24653\Desktop\carla-dev\carla-dev>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
Cloning into 'C:\Users\24653\Desktop\carla-dev\carla-dev\Unreal\CarlaUE4\Plugins\StreetMap'...
fatal: unable to access 'https://github.com/carla-simulator/StreetMap.git/': Failed to connect to 127.0.0.1 port 7890 after 2067 ms: Couldn't connect to server
系统找不到指定的路径。
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\Users\24653\Desktop\carla-dev\carla-dev\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
系统找不到指定的路径。
已复制         0 个文件。
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\" --zlib-install-dir "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Users\24653\Desktop\carla-dev\carla-dev\Build\" --generator "Visual Studio 16 2019"
**系统找不到指定的批处理标签 - already_build**

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
Util/BuildTools/Windows.mk:75: recipe for target 'setup' failed
make: *** [setup] Error 1

I downloaded the zips from GitHub manually and unzipped into the designated folders, however it reported "The system cannot find the specified batch tag". Can anyone help?

freefredx, 2024-09-24:
I've passed the zlib issue by changing the version to 1.13.1.But anyone who meet the similar problem no libpng?for me the default downloading address on sourceforge doesn't work.

MaharshiPathakCode, 2025-02-03:
> C:\Carla\carla>make PythonAPI -[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all -[Setup]: Asynchronous jobs: 8 -[Setup]: Boost toolset: msvc-14.2 -[Setup]: Install directory: "C:\Carla\carla\Build" -[Setup]: Installing zlib... -[install_zlib]: [Batch params]: --build-dir "C:\Carla\carla\Build" -[install_zlib]: A zlib installation already exists. -[install_zlib]: Delete "C:\Carla\carla\Build\zlib-install" if you want to force a rebuild. -[install_zlib]: Exiting... -[Setup]: Installing libpng... -[install_libpng]: [Batch params]: --build-dir "C:\Carla\carla\Build" --zlib-install-dir "C:\Carla\carla\Build\zlib-install" -[install_libpng]: A libpng installation already exists. -[install_libpng]: Delete "C:\Carla\carla\Build\libpng-1.2.37-install" if you want to force a rebuild. -[install_libpng]: Exiting... -[Setup]: Installing rpclib... -[install_rpclib]: [Batch params]: --build-dir "C:\Carla\carla\Build" -[install_rpclib]: Not cloning rpclib because already exists a folder called "rpclib-src". -[install_rpclib]: Generating build... CMake Error: The source directory "C:/Carla/carla/Build/rpclib-src" does not appear to contain CMakeLists.txt. Specify --help for usage, or press the help button on the CMake GUI.
> 
> ```
> -[install_rpclib]: [CMAKE ERROR] An error ocurred while executing the cmake.
> -[install_rpclib]: [CMAKE ERROR] Possible causes:
> -[install_rpclib]:                - Make sure "CMake" is installed.
> -[install_rpclib]:                - Make sure it is available on your Windows "path".
> -[install_rpclib]:                - CMake 3.9.0 or higher is required.
> -[install_rpclib]: Exiting with error...
> ```
> 
> -[Setup]: Ok, and error ocurred, don't panic! We have different platforms where you can find some help :)
> 
> * Make sure you have read the documentation:
>   http://carla.readthedocs.io/en/latest/how_to_build_on_windows/
> * If the problem persists, you can ask on our Github's "Building on Windows" issue:
>   [Building on Windows 0.8.X #21](https://github.com/carla-simulator/carla/issues/21)
> * Or just use our Discord channel!
>   We'll be glad to help you there :)
>   https://discord.gg/42KJdRj
>   make: *** [setup] Fehler 1
> 
> Can anybody help me with this problem?

aleksivu123, 2025-02-12:
> I have an issue with running prompt make PythonAPI,
> 
> Error: C:\Program Files\Epic Games\carla>make PythonAPI MakeFile build is currently disabled.

Same here. Any solutions?

himanshupradeep, 2025-02-12:
Hello !

Older solution doesn´t work anymore. CARLA changed the way to builds in version 0.10.0.
Please follow the instructions given in this new page.

https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/

Before starting with the steps mentioned in the above page link,

1. Uninstall Visual Studio 19 or 22 or both.
2. Old Python application
3. Old CMake application
4. Delete complete CARLA folder and Unreal Engine Folder

**git clone -b ue5-dev https://github.com/carla-simulator/carla.git CarlaUE5**
(The above command will clone the new CARLAUE5 to your local machine)

BEFORE YOU START THIS, kindly add the new environmental variable with this directory attached

Environment Variable - **CARLA_UNREAL_ENGINE_PATH**
Directory - **C:\CarlaUE5\Unreal\CarlaUnreal**

**cd CarlaUE5**  (Run in CMD)
**CarlaSetup.bat** 

Run these setup script and this will take care of installing all necessary dependencies. Later, follow the instructions on the page.
My build failed at certain stage but I am working on a solution. I will update once I have completed the installation. 

Regards
Himanshu