carla-simulator/carla #4860: Wrong car rendering

ZarkSPb, 2021-11-09:
Hi! I render some scenes, and on one image I see one car with right geometry and and another car with wrong geomenry.

 
[image]
[image]
[image]
[image]

How can I fix this? 

Carla 0.9.12
Windows 10
I rgb_create camera, set uo it in viewpoint, listen image and save it.

I try wait some world ticks and save only after 20 ticks without car moving. It makes the rendering (image) better, but not solve problem with car geometry.

Madecu, 2021-11-18:
Hi @ZarkSPb 
This is weird. What you are seeing is the LODs for the vehicles, simplified geometries that are rendered depending on the camera distance to improve performance. 

Can you show me how did you create your RGB camera? The vehicle should recognize that there is a camera near it and render the highest geometry model.

This is an example of how do we create an RGB camera from our blueprint library

`    ##  Create a client ##
    client = carla.Client('localhost', 2000)
    client.set_timeout(10.0)
    world = client.get_world()
    
    ##  Create the RGB camera from our Blueprint Library  ##
    blueprint_library = world.get_blueprint_library()
    camera_bp = blueprint_library.find('sensor.camera.rgb')
    camera_location = carla.Transform(carla.Location(x=-24.00, y=14.67, z=2.22), carla.Rotation(yaw=180))
    camera = world.spawn_actor(camera_bp, camera_location)
`

ZarkSPb, 2021-11-18:
Hi, @Madecu 

 I create RGB camera:
```
# Create a client
client = carla.Client('localhost', 2000)
client.set_timeout(20.0)
world = client.get_world()

# Create RGB camera
blueprint_library = world.get_blueprint_library()
camera_bp = blueprint_library.find('sensor.camera.rgb')
camera_bp.set_attribute('image_size_x', str(h_resolution))
camera_bp.set_attribute('image_size_y', str(v_resolution))
camera_bp.set_attribute('fov', str(fov_h))
camera_rgb = world.spawn_actor(camera_bp, spawn_points[0])
camera_rgb.listen(camera_listen)
actor_list.append(camera_rgb)
```
I run:
`start CarlaUE4.exe -quality-level=Epic -ResX=320 -ResY=240`

And  I noticed, when I use `no_rendering_mode=False` I get normal images, but it takes many-many world.tick (50 - 100 for high qualiti images, and this without car movind)

When I use `no_rendering_mode=True` I get weird car images (regardless of the numbers of world.tick), but surrounding area looks good.

yuvalHG, 2021-11-21:
I got the same issue.
My scene is dynamic, with moving cars and around 5 ego vehicles with RGB cameras and I can not pay 50-100 ticks to get a high quality image.
Is there any solution for getting well rendered car in synchronous mode?
I'm using Carla 9.11 on win10

closed 2023-06-02 by stale[bot]