carla-simulator/carla #4861: Unknown Socket Error

budrus123, 2021-11-09:
I have a very weird issue going on with Carla. I am running a reinforcement learning agent with CARLA and after a certain number of steps (varies each time) I get this error (shown in the image below). I have tried changing the socket, but that did not fix the issue at all. I also cannot find any logs or log files that can help me identify this issue. It is saying that an operation was attempted on something that is not a socket but I have no idea what this refers to. Any help would be greatly appreciated. 

[image]

For reference:
**CARLA version**: CARLA_0.9.12 (latest packeged build)
**Platform/OS**: Windows 11

budrus123, 2021-11-09:
Update on when this happens:

So after testing this happens when I have a sensor listen to data using the following code:
`self.collision_sensor.listen(lambda data: self.collision_history.append(data))`

This is in a function called `reset` which gets called thousands of times after the scenario is complete. But each time I call this function, I make sure to destroy the sensor and stop listening and then spawn and listen again. After some time I get the error above.

If I run my program without listening to the sensor, nothing crashes.

hh0rva1h, 2021-12-01:
We are experiencing exactly the same problem (see https://github.com/carla-simulator/carla/issues/4935) with Carla 0.9.13 under Linux. We are also doing reinforcement learning and are using the collision_sensor as well and destroy and respawn that sensor after each episode.
Also we notice that the segfault occurs always after an episode where a collision occurred that the end of the episode.
@MattRowe18 Basically my bug and this one describe the same issue, could you merge them somehow and make sure one person is assigned for both? Also please do not lable as "possible bug", this is hereby confirmed, happens to different users, on different platforms, this is a bug for sure.

hh0rva1h, 2022-02-16:
@Axel1092 @bernatx Please have a look, this issue makes us a lot of trouble, we have frequent crashes. Seems however also related to hardware, on some machines this happens very often on others hardly ever.

bernatx, 2022-02-17:
Hi, sure we will check this bug as soon as we get some time for it, asap hopefully. Thanks for the details. We have got some trouble also with the collision sensor, so something is there for sure. We are going to fix it, and keep you updated.

RocaPiedra, 2022-02-24:
I'm having the same issue when trying to stream the rgb sensor data to my visualization method. Do you have any workarounds? I haven't been able to test that part of the code, not even once.
What I'm doing is creating a "sensor_platform" in my main. That "sensor_platform" is defined in another python package and I perform the initialization of the carla client in the __init__ of that class. Could that be an issue?
I'm using Carla 0.9.13 in Windows.

lasuomela, 2022-02-28:
Encountering this bug also in 0.9.12. Docker image. This is super inconvenient.

hh0rva1h, 2022-03-15:
I thought about maybe it would help to recreate the client after each episode, however that leads to another problem: https://github.com/carla-simulator/carla/issues/3109
I am quite sure all those issues are related in the sense that socket handling between server / client is buggy.

arjunsk07, 2023-04-25:
@budrus123 Did you able to solve the error? the crashing of UE4.
I am also experiencing the same issue.
Can you please help me in this?

ruff-huettinger, 2023-05-08:
Also having this error. Cannot track it down, though.

ruff-huettinger, 2023-08-01:
For me, using the accelerometer instead of the collision sensor made things better. The error appears still from time to time.

Somdit, 2023-11-11:
I encountered the same issue on CARLA 9.14, and there seems to be no fixing for 2 years now. Also using a collision sensor for reinforcement learning that destroys and respawns the sensor when resetting the environment.

uthaypre, 2024-04-11:
I was able to solve this issue by adding a time-out and a world reload in my cleanup function which i call inbetween each reinforcement learning episode. This allowed me to still keep the collision sensor.

def cleanup(self):
   for sensor in self.world.get_actors().filter('*sensor*'):
      if sensor.is_alive:
         sensor.stop()
         sensor.destroy()

   for actor in self.world.get_actors().filter('*vehicle*'):
      if actor.is_alive:
         actor.destroy()

   self.client.set_timeout(4.0)
   self.world = self.client.reload_world()

soaham-pimparkar, 2025-02-19:
I am facing the same issue with Carla 0.9.15 on Windows 11. for some reason, my client.reload_world() method is sausing a different problems so can't use that work around either. @MattRowe18 @bernatx could you please look into this issue?

A lot more users might be facing this issue but prolly not being able to realize that the sensor callback is the reason behind the server crashing as mine gives no error / warning or pop up.

**ps:** other sensors are also giving the same problem :/ has to do something with the sensor callback after the sensor is manually destroyed