carla-simulator/carla #4862: Carla 0.9.9 Docker Image Build Error: Error 403 Forbidden

ucbmpc2021, 2021-11-10:
When I'm trying to build the docker image for carla branch 0.9.9, after running this command
docker build -t carla -f Carla.Dockerfile . --build-arg GIT_BRANCH=0.9.9

The error below will show:

I am on Ubuntu 18.04.

Just in case anyone encountered the same issue, I posted a solution here to solve this issue:

My fix was to change line 20 in MyBuildUtilsDocker.sh in the Docker container to:
`wget --user-agent="Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:94.0) Gecko/20100101 Firefox/94.0"  -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}"`
Here are the steps I followed:
1 - In Util/Docker, create a file called MyBuildUtilsDocker.sh
2 - cd ..
3 - run cat BuildTools/BuildUtilsDocker.sh > Docker/MyBuildUtilsDocker.sh
4 - Change line 20 in MyBuildUtilsDocker.sh to what I put above.
5 - cd Docker
6 - change Carla.Dockerfile to:

`FROM carla-prerequisites:latest

ARG GIT_BRANCH

USER ue4

RUN cd /home/ue4 && \
  if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git; \
  else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi && \
  cd /home/ue4/carla && \
  ./Update.sh && \
  make CarlaUE4Editor && \
  make PythonAPI
WORKDIR /home/ue4/carla
COPY MyBuildUtilsDocker.sh .  
RUN cat MyBuildUtilsDocker.sh > Util/BuildTools/BuildUtilsDocker.sh
RUN make build.utils
RUN make package
RUN rm -r /home/ue4/carla/Dist

WORKDIR /home/ue4/carla`

7 - Run docker build -t carla -f Carla.Dockerfile .

danielaDESM, 2021-11-12:
I have the same issue but with the latest version 0.9.12. But I have checked the link and it points correctly to the tar.gz that wants to download. So I do not understand this Error 403 FORBIDDEN. 

[image]

What exactly this instruction does?
wget --user-agent="Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:94.0) Gecko/20100101 Firefox/94.0" -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}"

ucbmpc2021, 2021-11-17:
Based on your screenshot this is the exact same problem as I had. Looks like 0.9.12 hasn't fixed this problem yet. You can try my solution as see if that fixes the problem :)

ucbmpc2021, 2021-11-17:
> I have the same issue but with the latest version 0.9.12. But I have checked the link and it points correctly to the tar.gz that wants to download. So I do not understand this Error 403 FORBIDDEN. [...]

Based on your screenshot this is the exact same problem as I had. Looks like even 0.9.12 hasn't fixed this problem yet. You can try my solution as see if that fixes the problem :)

Eddy-M-K, 2021-12-06:
I had the same issue on version 0.9.13 and this worked. Thank you @ucbmpc2021.

ucbmpc2021, 2022-01-06:
> I had the same issue on version 0.9.13 and this worked. Thank you @ucbmpc2021.

Sure thing. Happy to help!

closed 2022-01-06 by ucbmpc2021

coutyou, 2022-03-09:
Thanks, bro! it helps a lot :)

cmaboeker, 2022-07-22:
I ran into the same issue. Thank you for the fix @ucbmpc2021.

Bonifatius94, 2022-08-05:
Thanks @Inglam, helped me a lot.

Btw, why did noone patch the script. I mean the Dockerfile doesn't build. How do they even publish the image on DockerHub?! :sweat_smile:

cakiva, 2022-11-17:
Thanks @ucbmpc2021 ! (works for me on 0.9.13)
BTW, you can just add the following line to your Carla.Dockerfile:
`sed -i 's,wget -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}",wget --user-agent="Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:94.0) Gecko/20100101 Firefox/94.0" -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}",g' test/carla/Util/BuildTools/BuildUtilsDocker.sh`