carla-simulator/carla #4864: how to make CARLA Standalone

koochak-mehdi, 2021-11-10:
Hi,
I'm trying to build Carla from the source an get a standalone version of my changes. I've deleted all cars in a parking of Town05 in order to fill the slots randomly.
If I understood it correctly, after my change I can either get the executable directly from unreal engine or do `make package` command.
But in both cases I face the same problem, and there are missed lots of components. For example:
`LogInit: Display: LogCook: Warning: Unable to find package for cooking /Carla/carla/Unreal/CarlaUE4/Content/Carla/Static/GenericMaterials/Licenseplates/Textures/2TKS587`
I want to to how can I build properly a standalone version of my work.
for example I get this error while trying `make package`:
```
  LogInit: Display: NOTE: Only first 50 warnings displayed.
  LogInit: Display:
  LogInit: Display:
  1001LogInit: Display: Failure - 19119 error(s), 19252 warning(s)
  LogInit: Display: Failure - 19119 error(s), 19252 warning(s)
  LogInit: Display:
  Execution of commandlet took:  32.17 seconds
  LogInit: Display:
  Execution of commandlet took:  32.17 seconds
  LogShaderCompilers: Display: Shaders left to compile 0
  LogHttp: Display: cleaning up 0 outstanding Http requests.
  LogContentStreaming: Display: There are 1 unreleased StreamingManagers
  Exiting abnormally (error code: 1)
Took 98.771525s to run UE4Editor, ExitCode=1
ERROR: Cook failed.
       (see /home/mehdi/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
make: *** [package] Error 25
```
just to mention I could do `make launch` and even play inside of UE4, and here you can find my configuration:
CARLA version: 0.9.12
Platform/OS: Ubuntu 18.04
RAM: 16GB
SWAP: 8GB (partition)
GPU: RTX2080 qmax
Steps to reproduce: `make PythonAPI && make launch && make plugins && make package`
Python: conda 3.7

here you can find the log file for `make package`:
[Log.txt](https://github.com/carla-simulator/carla/files/7511417/Log.txt)

ndeshp2s, 2022-07-19:
@koochak-mehdi did u manage to solve the issue?

NnaYelsel, 2023-03-06:
Hello, any answer to this issue?

Personally, I achieve make PythonAPI and make launch steps but how can I get the same folder system than those of the Release version of the code (in order to launch it in another PC) ?

chuanye-Wang, 2024-07-29:
any solution? Please let me know if you did

closed 2025-09-22 by Blyron