carla-simulator/carla #4871: Is it possible to send the message from simulink to carla?

lzhhh93, 2021-11-11:
Hi everyone,

As I want to evaluate my control algorithm in Carla, I probably need this from-simulink-to-carla interface.

aadishdnaik, 2021-11-11:
You could look into this blockset provided by Mathworks. It basically allows you to send and receive data to/from Unreal Engine - https://www.mathworks.com/matlabcentral/fileexchange/65966-vehicle-dynamics-blockset-interface-for-unreal-engine-4-projects

lzhhh93, 2021-11-12:
> @lzhhh93 You could look into this blockset provided by Mathworks. It basically allows you to send and receive data to/from Unreal Engine - https://www.mathworks.com/matlabcentral/fileexchange/65966-vehicle-dynamics-blockset-interface-for-unreal-engine-4-projects

Thanks a lot. i am looking into it. 

I have also thought about sending and receiving data through a Python API: https://de.mathworks.com/matlabcentral/fileexchange/77221-how-to-interact-with-simulink-using-python

closed 2021-11-30 by MattRowe18

lzhhh93, 2021-12-02:
To connect MATLAB with Carla:

Use matlab engine ( https://de.mathworks.com/help/matlab/matlab-engine-for-python.html ) to send and get data from Python API of Carla. 

small hint: it is not neccesary to start MATLAB Engine for Python, if we only want to connect shared MATLAB session to MATLAB Engine for Python. Starting MATLAB Engine for Python takes a lot computation (somehow).

exemple:

in a python script:

        `## make sure the following command is executed in matlab command window, before running python scripts:
        ## matlab.engine.shareEngine
        import matlab.engine
        eng = matlab.engine.connect_matlab()
        throttle = eng.LongPID(self._k_p,error,self._k_d,_de,self._k_i,_ie)`

While in MATLAB:
       `function [throttle] = LongPID(k_p,error,k_d,de,k_i,ie)
               throttle =  (k_p * error) + (k_d * de) + (k_i * ie);
       end`